betaflight/betaflight #5159: TINYBEEF3 WONT FORCE BIND MODE FROM THE CLI OR COMMUNICATE WITH PRE BOUND DSMX

CRYESSO, 2018-02-11:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide information about your flightcontroller and other components including how they are connected/wired;
4. Add the used configuration and firmware version;
5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6. Remove this Text :).

CRYESSO, 2018-02-11:
USING ORANGE RX. SHOWS 3 UARTS IN BETA FLIGHT, BUT THE BOARD ONLY HAS ONE UART. SO TRIED WITH UART1 ON SERIAL, DIDNT WORK, SO TRIED WITH ALL THREE ON SERIAL. STILL NOTHING. SET UP ALL THE NECESSARY CONFIGS. TRIED DSM2 AND DSMX CONFIGS. WENT INTO CLI, SET SPEK BIND TO 3,5,7,8,9 ALL TO NO AVAIL. TRIED WITH AUTORESET ON AND OFF FOR 3,5,7,8,9 AND STILL NOTHING. NOT SURE WHAT TO DO NEXT. ALSO WENT TURN OFF SBUS INVERSION BUT THAT DIDN'T CHANGE ANYTHING. HELP!!!

spatzengr, 2018-02-11:
@CRYESSO, go to the above links for help, please.  As far as I'm aware, most of the Orange RXs are PWM or PPM outputs (not serial), so you have to use the PWM or PPM inputs on your FC board (not Uarts).  Someone will help you at the links above.
(https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech)

closed 2018-02-11 by mikeller