betaflight/betaflight #5164: Runaway prevention should be disabled when crash-flip mode is active

bkeinert, 2018-02-12:
Pretty much simply what the title says. Why?
Because runaway prevention tends to hinder crash-flip mode (I observed that you can only do one crash-flip "movement"). This even more prominent with PR #5163 - where a yaw movement pretty much directly disarms in crash-flip mode.

Would it be dangerous to remove runaway prevention when crash-flip mode is active? I don't think so. Discuss.

TheIsotopes, 2018-02-12:
imho i have also this problem on the bench ... i'm armed, moved the throttle stick in any position ... now if i'm using yaw, pitch or roll full stick input it goes in directly disarm ... is this the same what you mean ... disabled runaway prevention brings not this problem

mikeller, 2018-02-12:
@TheIsotopes: That is correct. For bench testing, runaway prevention needs to be disabled. I am wondering if we should add a facility for the configurator to do this temporarily (when it is connected), to keep users from disabling it permanently for bench testing, and then forgetting to re-enable it?

etracer65, 2018-02-12:
@bkeinert I'm not arguing that runaway takeoff prevention shouldn't be disabled in crash-flip mode, but I haven't seen the affect you're describing.  I've been doing a lot of indoor flying and have literally done dozens of crash-flips and never had runaway takeoff trigger.

It might be partially because of the way you're testing.  I suspect you're arming the quad upside down and testing crash-flip.  In this case runaway takeoff is still active because you were never in real flight to have it deactivate.  But in real use cases the prevention would have been deactivated already and thus couldn't trigger.

@TheIsotopes Yes, runaway takeoff prevention will cause this disarm because the quad is not responding to the PID controller (because no props) causing the pidSUM to rapidly increase - looking like a runaway takeoff event.

@mikeller Disabling temporarily while connected to the configurator might be a good idea.

In regards to crash-flip: the bigger question I have is why is the PID controller even active during crash-flip?  If the PID controller wasn't calculating pidSum then runaway takeoff couldn't possibly trigger.  Also the PID controller is running when disarmed - not sure the value in that (other than showing more accurate CPU load I guess).

However if desired it doesn't looks like it would be anything difficult to disable runaway takeoff prevention when in crash-flip mode.

etracer65, 2018-02-12:
And to follow-up on the bench testing:

I think having an "Enable Bench Testing" button at the top of the configurator would be a good idea.  This could popup a confirmation that the user confirms that the props are removed before continuing.  Until the user enters bench-test mode arming and motor testing would be disabled.  Once in bench-test mode runaway takeoff could be temporarily disabled as well.

mikeller, 2018-02-12:
@etracer65: Re runaway-takeoff-prevention interfering with crash-flip, good point, since runaway-takeoff-prevention is disabled **until power loss** after the first successful takeoff.

Re configurator, we already have this, including all the warnings about the bad things that are going to happen if the user does not remove the props, in the motor tab, for the 'enable motors' slider. This is currently also used to disable the arming prevention that is active when connected in configurator. I think adding it there, as a flag on the MSP command that disables the arming prevention, should do:

https://github.com/betaflight/betaflight/blob/040c46da079112896dc1ad94d3298d8b228b89c2/src/main/interface/msp.c#L1777-L1786

Do you have time to come up with a pull request for this?

bkeinert, 2018-02-12:
@etracer65 True. But: I sometimes like to use starting upside down as a handicap when flying races with my workmate. :) Well, yeah - I can just disable that feature if required.

etracer65, 2018-02-12:
@mikeller I have the time, but I'm not sure it will be that simple.  The switch on the motor tab just unsets the `MSP` arming disabled reason.  So how can I differentiate between it being turned off by the switch vs. not ever being enabled because the quad isn't connected to the configurator?  There is no persistence that we are in the configurator and MSP arming disabled was ever set.

I suppose I'll need to capture a static flag like `static bool armingDisabledMSPEver` or something like that.

etracer65, 2018-02-12:
Or more generically, a `bool connectedToConfigurator()` function.  Not sure I know enough about MSP handling to take that on though.

mikeller, 2018-02-12:
@etracer65: I'd introduce a boolean variable for 'runaway-takeoff-prevention temporarily disabled', and use an additional flag in the `MSP_SET_ARMING_DISABLED` to switch this variable on or off. I.e. when connecting, sending `MSP_SET_ARMING_DISABLED` with 1 (set arming disabled), 0 will disable arming. When enabling motor output, sending 0 (unset arming disabled), 1 (disable runaway-takeoff-prevention) will enable arming and inhibit runaway-takeoff-prevention, later on sending the command with 1, 0 will revert to arming disabled, runaway-takeoff-prevention off. On disconnect, 0, 0 will unset arming disabled, leaving runaway-takeoff-prevention on.

etracer65, 2018-02-12:
Created https://github.com/betaflight/betaflight/pull/5172 to allow bench-testing while connected to the configurator without triggering runaway takeoff.

etracer65, 2018-02-13:
Created https://github.com/betaflight/betaflight/pull/5176 to disable runaway takeoff when crash-flip is active.

mikeller, 2018-02-18:
@etracer65: Can this be closed?

closed 2018-02-18 by bkeinert