betaflight/betaflight #5168: TINYBEEF3 TARGET UART COUNT INCORRECT

CRYESSO, 2018-02-12:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide information about your flightcontroller and other components including how they are connected/wired;
4. Add the used configuration and firmware version;
5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6. Remove this Text :).

mikeller, 2018-02-12:
No description.

CRYESSO, 2018-02-12:
THE BOARD HAS ONE UART PORT. UNDER THE PORTS TAB, BETAFLIGHT HAS THE USB AND 3 UART PORTS.  TO HOOK UP A RECEIVER TO THIS PARTICULAR BOARD YOU HAVE TO PLUG INTO THE UART PORT AND CONFIGURE ONLY UART2 IN BETAFLIGHT. MY ISSUE IS IF YOU DON'T KNOW THAT TO START IT CAN BE VERY CONFUSING FOR WHAT IS SUPPOSED TO BE A SIMPLE AND INTUITIVE PLATFORM. i SIMPLY SUGGEST THAT THE PORT TAB REFLECT THE ACTUAL HARDWARE.

mikeller, 2018-02-12:
@CRYESSO: You write in mostly capital letters. This is generally seen as shouting, and rude. Try to avoid it in the future.

Also, please avoid opening multiple issues for the same problem.

Re TINYBEEF3, can you please try this and see if it fixes the problem:

[betaflight_3.3.0_TINYBEEF3.zip](https://github.com/betaflight/betaflight/files/1716317/betaflight_3.3.0_TINYBEEF3.zip)

CRYESSO, 2018-02-12:
Ok, so the new hex fixed the ports tab. Works great. Thanks for fixing
that. I know it is minor, but the next guy will have less trouble than I
did with initial configs.
Sorry about the duplicates earlier was trying to type the description when
you closed the first one. Didn't mean anything by the caps, was just being
lazy. Looked better than all lower case. :)
Thanks again, awesome work! HAVE A GREAT DAY!!!

PR #5170 (2018-02-12): Fixed UART count for TINYBEEF3.
Fixes #5168.
closed 2018-02-12 by mikeller