betaflight/betaflight #5169: TINYBEEF3 UART COUNT INCORRECT

CRYESSO, 2018-02-12:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide information about your flightcontroller and other components including how they are connected/wired;
4. Add the used configuration and firmware version;
5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6. Remove this Text :).

CRYESSO, 2018-02-12:
THE BOARD HAS ONE UART PORT. UNDER THE PORTS TAB, BETAFLIGHT HAS THE USB AND 3 UART PORTS. TO HOOK UP A RECEIVER TO THIS PARTICULAR BOARD YOU HAVE TO PLUG INTO THE UART PORT AND CONFIGURE ONLY UART2 IN BETAFLIGHT. MY ISSUE IS IF YOU DON'T KNOW THAT TO START IT CAN BE VERY CONFUSING FOR WHAT IS SUPPOSED TO BE A SIMPLE AND INTUITIVE PLATFORM. i SIMPLY SUGGEST THAT THE PORT TAB REFLECT THE ACTUAL HARDWARE.

mikeller, 2018-02-12:
Duplicate of #5168 5168.

closed 2018-02-12 by mikeller