betaflight/betaflight #5171: Gyro Overflow Detect false triggering

etracer65, 2018-02-12:
@martinbudden  I've been seeing some reports of apparent false triggering of the gyro overflow detection and have come to the conclusion that triggering off of the raw gyro data is probably not the best approach.  For example, here's a screenshot and log from a quad experiencing "yaw twitch" from the gyro when running in 32K mode.  The twitches are significant enough to trigger the overflow detection but are filtered out so that they aren't visible in the filtered gyro and aren't affecting flight.  Technically it's not "false" triggering based on the data but clearly not the goal of the feature.

[image]

[OverFlow On Yaw.BBL.zip](https://github.com/betaflight/betaflight/files/1716307/OverFlow.On.Yaw.BBL.zip)

This is clearly horrible gyro noise and the quad has problems, but that's a separate issue.

There were also some occurrences where gyro rates were only approaching 1500 deg/sec yet overflow detection was triggered.  I wasn't able to see raw gyro data in these cases but I'm suspecting that there were momentary spike in the raw gyro data causing the trigger.

I would suggest the following options:

1. Move the triggering out of the gyro driver and use filtered gyro data rather than raw.  I feel like this is the best approach.

2. Add a triggering delay on the raw data where the rate has to be above the threshold for X amount of time.  Not as thrilled about this option as it's just basically a lowpass filter and we can get the same with option 1.  Also concerned that we'd have to "guess" at an appropriate time delay.

The false triggering is having the negative affect of causing people to disable the gyro overflow detect through word of mouth - leaving people without YSTTM prevention.

ctzsnooze, 2018-02-19:
This quad needs a series filter in the gyro.  If people want to suddenly try 32k mode, with the internal gyro hardware lowpass at 8133Hz, instead of 8k mode when it is 256Hz, what do they expect?

If such a 'Kalman' concept series gyro filtering method is enabled, then perhaps the anti YSTTM detection should come after that initial filter (as if it was a more common 256k filter, which would be the case usually).

I am getting lots of fast spins on clipping things but since the anti-YSTTM code has been in place, I've not had any bad ones.  So we know it works with gyro data that comes in after a 256Hz hardware filter.

etracer65, 2018-02-19:
I'm close to submitting a pull request to shift the overflow detection/reset to using gyro data after the filtering.  This seems to resolve the problem.

On a separate note since I exclusively fly MPU6000 gyros I've disabled the overflow_gyro_detect as I don't like the behavior when clipping a gate.  Since the MPU6000 has no overflow problems it can react immediately to counter the yaw spin vs. the overflow method of waiting for it to settle down aerodynamically.  Since it's the winter for me I'm flying indoors in a vertically constrained area and the quicker recovery is needed to stay flying. Unfortunately with ICM-206xx gyros there's no option but to have the feature enabled (unless you want YSTTM events!).  Even though I know there have been reports of YSTTM-like events with MPU6000 gyros (possibly through multi-axis coupling as a theory), I've personally never experienced it and I don't believe we've seen any logs supporting it.

ctzsnooze, 2018-02-19:
I've had it happen on my MPU6000's several times, but not since anti-YSTTM was implemented.  It happens with a complex multi-axis spin where the PIDs get over-active in attempting to control things, and it just perpetuates for a long time with the motors out of phase with the rotation somehow.  There's no reversal but you still can get flyways.

It's true that only the ICM gyros need the anti overflow-sign-reversal code, but the anti-YSTTM code that cuts PIDs when they exceed a high threshold  is, I think, very useful, even for MPU6000's.  It has helped me after quite a few hard hits.  I sometimes get bad yaw spins after hitting gates, I just keep enough throttle on, and it typically settles down after a while then suddenly stops spinning and you're back in control.  If you haven't had that happen, you've not hit the gates the *right way*  :-)

etracer65, 2018-02-19:
I would love to see a log with a YSTTM-like event on a MPU6000 so we could really see chat's going on.  Might be a better way to recover.

It's really unfortunate that the overflow bug in the ICM-206xx gyros has put us in this situation where we need to limit the response from the flight controller during recovery.  I can really feel the difference in recovery from gate clips with the gyro overflow detect logic on vs. off.  I'm mainly lamenting the fact that the overflow bug has negatively impacted gyros that don't need the protection.  And yes I hit plenty of gates the _right way_! :)

As an example from this past Saturday, here's a short clip of a gate clip and recovery in an indoor event.  Gyro overflow detect is off.  It's amazing how fast it recovers.  I ultimately ended up crashing because I overreacted and and couldn't get level fast enough, but the PID controller did it's job.  And this is a height-constrained indoor field where getting above 30 feet and you're in the net so popping up during the spin would be a problem.  Also with gyro overflow detect on I would have likely been into the back wall behind the gate when the PID's were disabled and slowing down aerodynamically and the quad was in a ballistic trajectory.

https://youtu.be/CsVplklnYpk

ctzsnooze, 2018-02-19:
Recovery from a clean, low-rate yaw spin is going to be fast and effective.

But sometimes it isn't.  Then ,the recovery PID stuff fails and the motors go crazy, motor traces all moving between zero and 100%, and the quad climbs.  At the races last weekend even with anti-YSTTM code I was able to clip gates at speed and regularly climb, with no user input, maybe 20m in half a second.  In every case the climb stopped by itself, but not as quickly as I would like.  These were spins rates that were too low to trigger the anti-YSTTM code.  I occasionally trigger the YSTTM protection, if I do it really helps.

I have videos of these and logs in the earlier discussions on this topic.

If my race quad had done exactly that clip, the motors would have put so much effort into preventing the spin that I'd have hit the roof.

It's really powerful quads with massive motor authority that have the problem.  No-one can explain it otherwise.  For me the YSTTM issue came in around version 3.x of betaflight, before then recoveries were generally quick and uneventful.  But around that time my quads got lighter, faster, and less heavily filtered, and I moved to reversed prop rotation.

My son is a very hard racer but flies super heavy filtering (e.g. LPF's at 50Hz) due to a flexy frame and has normal prop rotation.  His quads are light, powerful and fast, but don't 'over-react' on clipping gates.  He hit quite a few but the quad just sort of spun briefly then got back to normal without drama.  In contrast, I fly latest code, minimal filtering, hard frame, reversed prop rotation though not high PIDs (P35 D14) because it is really a very very responsive light quad.  And if I clip something, particularly on diagonal angles so it's not just a clean yaw spin, the quad takes off and climbs a lot, with the correction often taking more time than expected.  These climbs are typically during multi-axis spins, not flat spins.

This behaviour is exactly the same whether ICM206xx or 6000 gyros, and I don't run 32k mode, all are simple 8k8k on the F4 ICM's and 4k4k on the MPU6000's, otherwise same code and same PIDs etc.

On that basis I don't think they are hardware related but I do think it has to do with filtering, such that in some situations the spin rate gets so fast that the quad reacts to every little angular change but somehow the response to those rotations becomes out of phase somehow.  Once you get multiple motors hitting 100%, air mode then actively cuts any that are not 100%, and there's a bit of chaos going on.

If you've got a quad that does respond 'nicely', just disable the anti-YSTTM code, you don't need it.  

I don't mind if the default for anti-YSTTM is on or off.  

On the other hand, the anti-reversal code for ICM gyros in 32k mode is essential, and should always be on in 32k mode; it doesn't trigger until a much higher point, and all it does is keep the sign correct.  It does not limit recovery in any way.  Again even for these gyros, the secondary anti-YSTTM code is optional; if you had a heavy slow quad and didn't crash hard you don't need it.

etracer65, 2018-02-19:
The slew filter is enabled if gyro_overflow_check = OFF.  The problem is that the slew filter doesn't work unless the gyro is in 32K mode which was the reason for the overflow detection code.  So anybody with an ICM gyro really needs to have the gyro_overflow_detect = ON unless they know what they're doing and properly running in 32K mode with the high-frequency DLPF in the gyro.

Ideally we would only default the gyro_overflow_check = ON for ICM gyros, but unfortunately the gyro detection code is a bit of a mess and is target-based rather than determined at run time.  Some targets have multiple gyro options (like AIRBOTF4 like I fly) so we have to err on the side of enabling it since the FC *could* have an ICM gyro.  So maybe in the future we'll clean up the gyro detection code and make it better.

etracer65, 2018-02-23:
@Ozzy33 I've finished the changes to use filtered gyro data instead of raw for the gyro overflow detect.  Can you try this version to see if it helps with the false triggering you were experiencing?

[betaflight_3.3.0_REVOLT.hex.zip](https://github.com/betaflight/betaflight/files/1749996/betaflight_3.3.0_REVOLT.hex.zip)

Ozzy33, 2018-02-23:
Sure, will load it tonight and fly in the morning.
I should use set 
debug_mode = RUNAWAY_TAKEOFF and 
gyro_overflow_detect = YAW, correct, anything else?
Thanks again for everybody's time and effort!

etracer65, 2018-02-23:
That's fine.  We don't need to worry about the raw gyro anymore so no need to log it.  Also I don't expect runaway takeoff to have any part in this but that won't hurt anything.

Ozzy33, 2018-02-23:
I have tried the updated code with filtered data and is working much better (zero false triggers so far), thanks!
I had a couple of triggers do to to low thro flying (with in a few seconds of launching), so I adjusted these below and will fly at lunch.

set runaway_takeoff_deactivate_throttle_percent = 15
set airmode_start_throttle_percent = 15

My quad can hover at ~12% thro, do you think the above new setting are about right (to high or low)?

etracer65, 2018-02-23:
Probably a good starting point.  Might need to lower runaway takeoff.  You need to spend time above this limit to accumulate deactivation time.

Airmode you just to exceed once.

etracer65, 2018-02-23:
No, it hasn't been merged yet.  If you can post your testing feedback on the pull request that will help document the testing of the fixes.

https://github.com/betaflight/betaflight/pull/5284

closed 2018-02-24 by mikeller

etracer65, 2018-02-24:
@Ozzy33 The change was merged and is available in Jenkins build 634 or later.