betaflight/betaflight #5174: CLRACINGF4 LED not working with 3.3RC1

kicekilla, 2018-02-12:
I tried the latest 3.3 build last night on my new build using a JBF4 FC. after hooking up LEDs, I went to test them and they are flickering in a sort of rainbow pattern and every 4th LED is out. I thought it was the strip so I swapped it... same thing. 

Flashed back to 3.2.5 and LED's work as assigned (green larson scanner). I will post a pic of them working and not. I tried changing LED inversion in the CLI 0-7 on 3.3RC1 and no dice.

mikeller, 2018-02-12:
I tested 3.3.0-RC1 last night with my SPRACINGF3EVO (clone) brushed micro, and the LEDSTRIP was working as expected. It stands to reason that this is caused by some sort of (DMA?) conflict, potentially with Dshot. What's your `diff`, @kicekilla?

mikeller, 2018-02-13:
@kicekilla: Have you tried resetting your config to defaults, and then restoring it?

kicekilla, 2018-02-13:
Sitting down to play now.. I will try resetting to defaults and check it out.. and post a diff

kicekilla, 2018-02-13:
Just tried switching to multishot and LED's work as they should. Going to try to reset defaults and see if that allows me to use dshot600

`# diff all

# version
# Betaflight / CLRACINGF4 (CLR4) 3.3.0 Feb 12 2018 / 13:08:53 (040c46da0) MSP API: 1.37

# reset configuration to default settings
defaults nosave

# name
name HOTSTART

# resources

# mixer

# servo

# servo mix

# feature
feature MOTOR_STOP
feature TELEMETRY
feature LED_STRIP
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE

# map
map TAER1234

# serial
serial 2 64 115200 57600 0 115200
serial 3 8192 115200 57600 0 115200

# led
led 0 0,0::CO:6
led 1 1,0::CO:6
led 2 2,0::CO:6
led 3 3,0::CO:6
led 4 4,0::CO:6
led 5 5,0::CO:6
led 6 6,0::CO:6
led 7 7,0::CO:6
led 8 8,0::CO:6
led 9 9,0::CO:6
led 10 10,0::CO:6
led 11 11,0::CO:6
led 12 12,0::CO:6
led 13 13,0::CO:6
led 14 14,0::CO:6
led 15 15,0::CO:6
led 16 15,1::CO:6
led 17 15,2::CO:6
led 18 15,3::CO:6
led 19 15,4::CO:6
led 20 15,5::CO:6
led 21 15,6::CO:6

# color

# mode_color
mode_color 6 0 2
mode_color 6 1 1

# aux
aux 0 0 1 900 1050 0
aux 1 13 4 1300 1700 0
aux 2 15 0 1300 1700 0
aux 3 28 0 1300 2100 0
aux 4 36 2 900 1100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set rssi_channel = 8
set serialrx_provider = FPORT
set blackbox_p_ratio = 128
set min_throttle = 1060
set max_throttle = 1984
set motor_pwm_protocol = DSHOT600
set beeper_dshot_beacon_tone = 2
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 1
set osd_warnings = 63
set osd_rssi_alarm = 40
set osd_cap_alarm = 1000
set osd_rssi_pos = 2074
set osd_tim_2_pos = 2455
set osd_vtx_channel_pos = 2400
set osd_current_pos = 2048
set osd_mah_drawn_pos = 2080
set osd_craft_name_pos = 2432
set osd_warnings_pos = 2090
set osd_avg_cell_voltage_pos = 2060
set osd_stat_max_spd = OFF
set vtx_band = 3
set vtx_channel = 4
set vtx_power = 2
set vtx_low_power_disarm = ON
set vtx_freq = 5645

# profile
profile 0

set dterm_lowpass_type = PT1

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 2

# save configuration
save
# `

kicekilla, 2018-02-13:
No change when resetting defaults and re-flashing and re-configuring from scratch. 

testing motors in motors tab works as it should on dshot and multishot. If I arm to test motors in dshot600 motors 3 and 4 do not spin. using multishot, they all spin as they should. 

looks as though it is a dshot issue? 

My setup:
JBF4 AIO flashed with CLRACINGF4 3.3RC1
spedix ES30HV blheli_S (reflashed with 16.67 dshotcommands)

Should I rename the title as CLRACINGF4  Dshot DMA bug?

mikeller, 2018-02-13:
We are closing in on the cause. Can you do a `dma list` and post the output please? Then do a `set dshot_burst = off`, try again if Dshot / LED_STRIP works, and post the output of `dma list` again.

kicekilla, 2018-02-13:
`# dma list

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: LED_STRIP
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
`

after set dshot_burst = off, all 4 motors spin as they should.. they sound bad, but I have some stuff loose so it might just be a weird vibration. LED strip also works with burst off. 

dma list with set dshot_burst = off

`Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: MOTOR 4
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: MOTOR 3
DMA1 Stream 7: LED_STRIP
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: MOTOR 2
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: MOTOR 1
DMA2 Stream 7: FREE

# `

kicekilla, 2018-02-13:
Let me know if you need me to run anything else... I am off to bed for the night.  Thanks for your hard work : )

mikeller, 2018-02-13:
All good, problem identified.

kicekilla, 2018-02-14:
so is this safe to fly with dshot_burst=off? or should I flash to the latest 3.3 nightly?

mikeller, 2018-02-14:
It's entirely safe - the change that I made in #5180 just changes the default value for `dshot_burst` from on to off.

kicekilla, 2018-02-14:
awesome, just wanted to check! can't wait to maiden tomorrow. I'll report if I find anything else.

closed 2018-02-14 by mikeller

Martivip, 2023-07-20:
Thansk to @V-22 on Discord the solution was `dma pin C08 1` but the downside of that is lack of Analog OSD, not an issue for digital setups 

# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: DSHOT_BITBANG 2
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: LED_STRIP
DMA2 Stream 5: FREE
DMA2 Stream 6: DSHOT_BITBANG 1
DMA2 Stream 7: FREE

PR #5180 (2018-02-14): Disabled Dshot DMAR for CLRACINGF4.
Fixes #5174.