betaflight/betaflight #5177: Proposal: Remove duplication between BOX3DDISABLE / BOX3DONASWITCH

mikeller, 2018-02-13:
I just had a thought (thanks @teracis for bringing it up):

It should be possible to merge the functions of the `BOX3DDISABLE` and `BOX3DONASWITCH` switches, and replace it with a `switched_3d` (or similar) parameter:

`switched_3d = off`:
`BOX3D` switches between 'normal' 3D (idle mid stick) and non-3D.

`switched_3d = on`:
`BOX3D` switches between normal and inverted 3D.

Thoughts, @brycedjohnson?

brycedjohnson, 2018-02-13:
Yeah that sounds like a good idea.  That would actually save some potential weirdness if someone had 3DDISABLE and 3DONASWITCH on at the same time.

mikeller, 2018-02-13:
Do you have time to come up with a pull request for this?

brycedjohnson, 2018-02-13:
I'm pretty booked up at the moment - I just moved and working on unboxing and setting up the drone lab again....

If I did have some time I have #4796 on my todo list to get fixed asap.

mikeller, 2018-02-13:
Good point. I'll look into it then.

PR #5179 (2018-02-17): Unified `BOX3DDISABLE` and `BOX3DONASWITCH` switches.
This saves one switch, and removes duplicity between `BOX3DDISABLE` and `BOX3DONASWITCH`, which could never be used together in a meaningful way anyway.

@brycedjohnson: Does this look good to you?

(I have inverted the way the switch works for '3D on a switch', so that the 'on' position of the switch results in positive thrust min -> max range for both 'normal' and '3D on a switch'.)

@ledvinap: Can you please check if this is the right way to remove a box?

Fixes #5177.
closed 2018-02-17 by mikeller