betaflight/betaflight #5184: RSSI scaling for Graupner HoTT receivers with SUMD

BlackPixxel, 2018-02-13:
Graupner recently allowed their GR12L and GR12L SUMD+T receivers to output RSSI on channel 12 or channel 16.

This value goes from 1100 for no signal to 1900 for maximum signal strength. Since this does not use the full range of 1000 to 2000, it causes the RSSI to be read between %9 and 89% instead of 0% and 99%.

This could be fixed by either changing the RSSI scale when SUMD is selected, or by allowing the user to enter receiver-specific min and max values for RSSI.

mikeller, 2018-02-13:
I was planning on adding generic scaling for RSSI anyway, this should cover this use case.

BlackPixxel, 2018-02-14:
Yes, that would be even more flexible!

Lord-Quake, 2018-03-23:
Is this request still open?
I'm in the same boat albeit I'm using INAV 1.9.0.

mikeller, 2018-03-24:
@Lord-Quake: It's still on my to-do list.

Lord-Quake, 2018-05-31:
I've just uploaded build 884 since I'm using HOTT and SUMD with RSSI.
What is the CLI command(s) so that I can change the ranges from 1100 - 1900 to 1000 - 2000?

mikeller, 2018-05-31:
@Lord-Quake: GitHub is for questions and discussion around the **development** of the firmware and tools. For questions around the configuration and use of Betaflight, please visit [RCGroups](https://www.rcgroups.com/forums/search.php?query=betaflight%20cleanflight%20configuration%20use).

Lord-Quake, 2018-05-31:
My sincere apologies!

https://www.rcgroups.com/forums/showpost.php?p=39733201&postcount=55210

Lord-Quake, 2018-06-01:
No answers to my question on RCGroups.
Betaflight on Facebook could not help.

Examples:
set rssi_scale = 90 -> 1100 = 9% | 1900 = 81%
set rssi_scale = 110 -> 1100 = 11% | 1900 = 99%

As far as I can tell this commit is not working for Graupner SUMD and RSSI.

mikeller, 2018-06-01:
@Lord-Quake:

> set rssi_scale = 90 -> 1100 = 9% | 1900 = 81%
> set rssi_scale = 110 -> 1100 = 11% | 1900 = 99%

> As far as I can tell this commit is not working for Graupner SUMD and RSSI.

This looks like this feature does **exactly** what it is meant to do - it allows you to scale the RSSI value that is reported.

mikeller, 2018-06-02:
Ok, I had a look into the code. Looks like setting the lower / upper bounds of the RC input range is only supported for the first 4 channels - this is a different issue from scaling of the RSSI signal read from a channel. I'll look into making the bounds adjustable for all channels where this matters.

closed 2018-06-02 by mikeller