betaflight/betaflight #5186: why is my quad picture missing and cant get back... the one wiht motor icons

docjoens82, 2018-02-13:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide information about your flightcontroller and other components including how they are connected/wired;
4. Add the used configuration and firmware version;
5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6. Remove this Text :).

docjoens82, 2018-02-13:
ive went to view and set to quad clicked on and off on the airplane figure but can not get quad picture to show up... not sure where all settings are saved but something is wrong here deleted chrome , extensions , removed %AppData% folder of goggle and reinstalled same issue not sure what button i hit to make it go but cant get it back in cleanflight black log or beta

MiddleMan5, 2018-02-13:
Please read and follow the commenting guidelines, otherwise your issue will likely be ignored.