betaflight/betaflight #5187: Estimate heading from GPS for multirotors

mluessi, 2018-02-13:
Currently we need a compass to be able to correctly estimate the heading for multirotors. Heading is used to e.g. correctly show the home arrow in the OSD, which is nice for long range flights.

For airplanes, we use the GPS to estimate the heading, which makes sense as airplanes are always flying forward. However, the same is pretty much always true for FPV multirotors, especially long range ones which are just flying around and not do wild tricks ;). So, IMO it would make sense to also use the GPS for heading estimation in quads when no compass is available. This probably wouldn't work for RTH etc, but would be an improvement for displaying the home arrow.

Thoughts?

AirBreak69, 2018-02-13:
We are working on that topic. Pls look at #4800 and #4817.
It's almost ready to be published, but will take some weeks to get it into the next release.
You should close this - it's a duplicated feature request.

mluessi, 2018-02-13:
@AirBreak69 perfect, thanks!

closed 2018-02-13 by mluessi