betaflight/betaflight #5188: serialrx_inverted ON/OFF switched?

silenec, 2018-02-13:
I used sbus_inversion = OFF in BF 3.2.2, SBUS receiver worked.
Today I flashed BF 3.3.0 RC1 and the receiver stopped working with serialrx_inverted = OFF.
Then I tried serialrx_inverted = ON and the receiver works again. Not sure if this is a bug or not?

I use OMNIBUS F3.

```
# diff
# Betaflight / OMNIBUS (OMNI) 3.3.0 Feb 12 2018 / 13:25:31 (040c46da0) MSP API: 1.37
name QAV 210
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource CAMERA_CONTROL 1 B06

feature -RX_PPM
feature RX_SERIAL
feature MOTOR_STOP
feature ANTI_GRAVITY
feature DYNAMIC_FILTER
beeper -ON_USB
serial 0 2048 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
aux 0 0 0 1050 1250 0
aux 1 28 0 1125 1500 0
aux 2 2 0 900 1125 0
aux 3 13 1 1150 1250 0
aux 4 26 0 1050 1250 0
rxfail 3 h
set gyro_sync_denom = 2
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 0
set rssi_channel = 9
set max_aux_channels = 5
set serialrx_provider = SBUS
set min_throttle = 1085
set motor_pwm_protocol = DSHOT300
set motor_pwm_rate = 8000
set failsafe_delay = 10
set failsafe_throttle = 1200
set failsafe_throttle_low_delay = 200
set align_board_yaw = 90
set bat_capacity = 1500
set ibata_scale = 236
set ibatv_scale = 400
set ibatv_offset = 100
set beeper_dshot_beacon_tone = 2
set small_angle = 180
set auto_disarm_delay = 30
set deadband = 3
set yaw_deadband = 2
set pid_process_denom = 1
set osd_rssi_alarm = 40
set osd_cap_alarm = 1500
set osd_alt_alarm = 10000
set osd_vbat_pos = 385
set osd_rssi_pos = 2401
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2454
set osd_flymode_pos = 2081
set osd_throttle_pos = 2424
set osd_vtx_channel_pos = 377
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 2441
set osd_mah_drawn_pos = 2448
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 91
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 58
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 2391
set osd_pid_roll_pos = 0
set osd_pid_pitch_pos = 0
set osd_pid_yaw_pos = 0
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_avg_cell_voltage_pos = 2433
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2186
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_alt = ON
set osd_stat_endbatt = ON
set vcd_video_system = 2
set camera_control_key_delay = 150
set camera_control_internal_resistance = 500
profile 2

set dterm_lowpass_type = PT1
set anti_gravity_gain = 2000
set p_pitch = 70
set d_pitch = 22
set p_roll = 54
set d_roll = 20
rateprofile 0

set thr_mid = 0
set thr_expo = 50
set yaw_rc_rate = 110
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80

# 
```

mikeller, 2018-02-14:
@silenec: Thanks for bringing this up. This change has not been properly pointed out in the release notes:

The parameter `sbus_inversion` has been changed into `serialrx_inverted`, and the way it is applied has been changed as follows: It now applies to all RX protocols, not just SBus, and instead of switching the UART to normal when off and inverted when on, 'off' now means that the port is set to whatever the default is for the selected protocol (i.e. inverted for SBus, not inverted for SUMD), and 'on' means that the port is inverted from default (i.e. not inverted for SBus, inverted for SUMD).

This has now been updated:
https://github.com/betaflight/betaflight/releases/tag/v3.3.0-RC1

mikeller, 2018-02-18:
@silenec: Did this explanation help, and can this be closed?

silenec, 2018-02-18:
Yes, thank you.

closed 2018-02-18 by mikeller

BongSpore, 2022-06-09:
hi I just new to fpv and i build 5inch using speedybee im having a problem in betafligjt no receiver input , i conect rx to tx pad  ..im using ELrS module .it can bind to reciever.