betaflight/betaflight #5189: [BUG] KIWIF4V2 3.2.4 and 3.3RC1 - SDCARD causing intermittent issues

MiddleMan5, 2018-02-13:
## KIWIF4V2 3.2.4 and 3.3RC1 - SDCARD causing intermittent issues

## SYSTEM INFORMATION
**TARGET:** KIWI F4 V2 - Issue has been observed since 3.2.4
**RECEIVER:** XM+ on **UART1**
**ESCs:** 4 x BLHELI_S Q-H-20 Targets. Version 16.7 - BF configured for **DSHOT600**
**VIDTX:** TBS Unify Race HV to Battery Pads on PDB - **Smart Audio** on **UART3**
**WIRING:** Standard 4-motor recommended configuration with 5V LED strip connected to **VTX.DTA**

## OBSERVED ISSUES
- SD Card halts system boot intermittently
- System boots with SD Card disconnected, intermittent with SD inserted
- SD Card logging does not work, configurator reports "Fatal Error" in Blackbox tab
- OSD (Headset) will (intermittently) show distorted BF logo and broken text, requires full power cycle
- Configurator will not **re-connect** to target over USB, reports error with closing serial port

## EXTRA INFORMATION

### 3.3RC1 Diff
```
# version
# Betaflight / KIWIF4V2 (KIW2) 3.3.0 Feb 12 2018 / 13:19:47 (040c46da0) MSP API: 1.37

# name
name Chameleon

# feature
feature LED_STRIP
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -SYSTEM_INIT
beeper -ON_USB

# serial
serial 0 68 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200

# led
led 0 8,0::AO:2
led 1 7,0::AT:2
led 2 6,0::AT:5
led 3 5,0::AO:2

# mode_color
mode_color 6 0 3
mode_color 6 1 6

# aux
aux 0 0 3 1400 1600 0
aux 1 27 4 1925 2100 0
aux 2 13 1 1850 2100 0
aux 3 15 4 1400 1600 0
aux 4 26 2 900 1075 0
aux 5 35 9 1925 2100 0
aux 6 39 2 1925 2100 0

# master
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 0
set min_check = 1000
set max_check = 2000
set rssi_channel = 16
set max_aux_channels = 12
set blackbox_p_ratio = 64
set blackbox_device = SDCARD
set min_throttle = 1000
set dshot_idle_value = 200
set motor_pwm_protocol = DSHOT600
set current_meter = ADC
set vbat_scale = 58
set ibata_offset = 11
set small_angle = 180
set deadband = 1
set yaw_deadband = 1
set pid_process_denom = 1
set osd_tim2 = 1
set osd_vbat_pos = 2092
set osd_rssi_pos = 2088
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2103
set osd_flymode_pos = 333
set osd_throttle_pos = 2273
set osd_vtx_channel_pos = 2520
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 2496
set osd_mah_drawn_pos = 2466
set osd_craft_name_pos = 489
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2473
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83

# profile
profile 0

# rateprofile
rateprofile 0

set roll_srate = 81
set pitch_srate = 82
set yaw_srate = 89
```

MiddleMan5, 2018-02-14:
I'm not comfortable with calling this *fixed, but after trying a couple SD Cards, it finally started working. This issue needs better documentation, it would be really nice to know what was causing the strange behavior with an "incompatible SD Card".

Nonetheless, the FC is functioning again with a new SD card

closed 2018-02-14 by MiddleMan5

mikeller, 2018-02-14:
Agreed that this is not ideal. Not sure how much of this is caused by the SD cards being accessed in SPI mode by almost all targets. This mode isn't used much by other devices, so it's possible that card manufacturers are not concerned about not properly supporting it. This might be improved by SDIO support that has recently been added to Betaflight (#4854). The bad news is that this requires different wiring of the SD card slot, so it won't be available on almost all existing targets.