betaflight/betaflight #5190: [Wiki] Plum F4 specs verification

bbbenji, 2018-02-13:
As far as I know, the FlyingLemon Plum F4 does not have a current sensor. If someone can please verify this I can go ahead and submit a PR for a correction this page: https://github.com/betaflight/betaflight/wiki/Board---PLUMF4

mikeller, 2018-02-13:
According to the target definition it does have a current sensor - not sure if it is available on a pin on the PLUMF4 board. @flyinglemonfpv would be the authority to ask.

bbbenji, 2018-02-13:
Perhaps this is in conjunction with the Kiwi PDB which is supported and has a shunt resistor for current sensing.

https://flyinglemon.eu/flight-controllers/44-kiwi-pdb.html

MiddleMan5, 2018-02-13:
I believe @bbbenji is correct, the PLUM is a variant of the KIWIF4 which includes a current sensor on its dedicated PDB. It was probably an oversight to include the current sensor on the KIWI

jflyper, 2018-02-13:
It is not the "on-board ADC", but "on-board current sensor" that is not on-board.

MiddleMan5, 2018-02-13:
The current sensor **is** an ADC, the naming scheme here is ambiguous. Regardless, there is nothing on-board to handle current sensing conversion.

mikeller, 2018-02-18:
@MiddleMan5: Without getting into hair splitting, current sensing is done with a shunt / comparator to convert the current into a voltage, **and** an ADC to digitise said voltage. Half of this is on-board of the PLUMF4, the other half isn't.

Are there any suggestion for a fix / documentation update, or should this be closed?

bbbenji, 2018-02-19:
Can we agree that the Wiki should be updated to state `ADC current sensor input` instead of `Current sensor`?

mikeller, 2018-02-19:
@bbbenji: That sounds reasonable. Can you please do the update - the wiki is writable by all users.

bbbenji, 2018-02-19:
Resolved, thanks everyone.

closed 2018-02-19 by bbbenji