betaflight/betaflight #5197: wiki correcting in sonar.md

aster94, 2018-02-14:
hello,

this is very of little importance, but in the wiki i think there is an error in sonar.md https://github.com/martinbudden/betaflight/blob/master/docs/Sonar.md
there is wrote that is needed a resistor in the trigger and echo pin, but the echo is an output so it is not needed. While that is true for the trigger which is an input

Bye bye you are awesome!

mikeller, 2018-02-15:
What you are looking at is an older version of this page that isn't in the actual betaflight repository. The current page has been updated: https://github.com/betaflight/betaflight/blob/master/docs/Sonar.md

aster94, 2018-02-16:
You are right, and anyway it is not specified to put the resistor only in the trigger pin

mikeller, 2018-02-17:
Can you please close this, @aster94?

aster94, 2018-02-17:
Ok, were i wrong and the resistor is needed in both pins?

closed 2018-02-17 by aster94