betaflight/betaflight #5201: motor assignment presets wrong for "kisscc"-target

Mode-1, 2018-02-15:
3 of 4 motor assignments are wrong by default. Example is valid for "QUADX1234":
[image]
3 of 4 consequently wrong in all other variations too, like "QUADX" etc.
Spin directions with false mapped motors stay correct - so no chance for detecting
a wrong behavior before flight, except with a single motor test in the GUI (which i did - phew !).
My quad is tested in flight, native 1234 mapping with kiss-firmware - so any wiring fault can be excluded here in my setup. Kisscc is fitted in original 0° heading (arrow to the front), no upside down or something.

This is the defaults (Betaflight / KISSCC (KISS) 3.3.0 Feb 12 2018 / 13:18:22 (040c46da0) MSP API: 1.37)
resource MOTOR 1 B14
resource MOTOR 2 B00
resource MOTOR 3 B15
resource MOTOR 4 A08

This is how it works for "QUADX1234" and "QUADX" (should be fine then for others settings too)
resource MOTOR 1 A08
resource MOTOR 2 B00
resource MOTOR 3 B14
resource MOTOR 4 B15

I think it's an issue and it should be verified and updated for the new target "kisscc".

# diff
# Betaflight / KISSCC (KISS) 3.3.0 Feb 12 2018 / 13:18:22 (040c46da0) MSP API: 1.37
resource MOTOR 1 A08
resource MOTOR 3 B14
resource MOTOR 4 B15

feature TELEMETRY
map TAER1234
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
profile 0

rateprofile 0

ronlix, 2018-02-16:
Hi,

the BF normal mix (QUADX) workes correct, but the (QUADX1234) is wrong.

its:
BF GUI <-> CC board
1          <->        3
2          <->        2
3          <->        4
4          <->        1
tested with the motor tab with v3.3.0-RC1 

regards

Felix

Mode-1, 2018-02-16:
Thanks for testing/verifying Felix !
Was out flying today, tested a 250quad+KissCC and BF3.3.0-RC1 with stock pids. Had level, horizon and acro mode tested - all working flawlessly with the translation-assignment for QUADX1234 (resource MOTOR 1 A08,  resource MOTOR 3 B14, resource MOTOR 4 B15) which mirror your results. Thought all mixes were wrong when one mix is wrong - my fault, sorry. Anyway for the time being resource remapping is a good workaround for QUADX1234. cheers.

mikeller, 2018-02-17:
@Mode-1: Sorry for not picking up on this earlier. I assumed that you'd tried to wire up your quad as shown below (which is called 'Quad X 1234' for obvious reasons). If you want to use the 'default' motor order, then you need to select 'Quad X'.

[image]

Can you please close this?

Mode-1, 2018-02-17:
@mikeller
Any noob crossing over from Kiss to BF will wire his quad according to 1234 - pattern. It's native when you buy this pcb. Philip Seidel has the most common wiring diagram for KissCC:
https://blog.seidel-philipp.de/flyduino-kiss-compactctrl-cc-anschlussplan-connection-plan/
That's why someone flashing BF will end up like i did and will choose "Quad1234". Pretty dangerous in my opinion when he would get the devils-quad-spin instead....

mikeller, 2018-02-17:
@Mode-1:

Mode-1, 2018-02-17:
@mikeller 
yh i will test it - give me 2h pls :-)

Mode-1, 2018-02-17:
@mikeller 
- made a DFU backup of my working quad, then converted the .hex you supported in the .zip-file to .dfu
- flashed it to my quad, had a individual motor-tests
Result: all motors correctly assigned for "QuadX1234"
For your reference here's my (kiss-native) wiring:
[image]
Offtopic: was brave, put props on, cautiously tried to hover my quad using level mode, but it danced on the ground, so i stopped. Evaluated and found out you had custom (non-stock) filter settings which my quad does not work with. So for safety reasons: not involved readers please don't use this .zip for flight tests :-) Anyway - correct assignment confirmation - Thx mikeller.

mikeller, 2018-02-17:
@Mode-1: Thanks for testing. All of the settings in the build I gave you (which is from #5217) are default settings for 3.3. So it's probably some of your settings that you will have to update for 3.3, or some that were renamed and did not restore properly.

Mode-1, 2018-02-18:
You are 100% right mikeller - for some reason flashing the chip with STM-defuse did NOT erase it fully
and my old motor assigments "survived" somehow. Btw, i had 3.3.0-RC1 on it before.
Now it gets complicated, since the previous test was only ok because of my old settings, sorry.
Different approach: flashed my KissCC with BF-configurator, full chip erase.
"QuadX1234" is preset in the config now.
Did QuadX1234 motor tests and motor 1 is mapped to motor 3 etc... all wrong except motor 2.
So i have to apologize (!) - test failed here - i got to withdraw my last comment.
This is why it failed:
# dump
# Betaflight / KISSCC (KISS) 3.3.0 Feb 18 2018 / 03:09:29 (3741b9ec0) MSP API: 1.37
name -
resource BEEPER 1 B13
resource MOTOR 1 B15
resource MOTOR 2 B00
resource MOTOR 3 A08
resource MOTOR 4 B14

Works here only when it's set like in the photo above:
resource MOTOR 1 A08
resource MOTOR 2 B00
resource MOTOR 3 B14
resource MOTOR 4 B15

mikeller, 2018-02-18:
@Mode-1: No problem. the persistent resource assignments can be pesky when testing. I've shot myself in the foot in the same way a number of times in the past.
Can you please try this (again from #5217):

[betaflight_3.3.0_KISSCC.zip](https://github.com/betaflight/betaflight/files/1734336/betaflight_3.3.0_KISSCC.zip)

Mode-1, 2018-02-18:
Bullseye ! Worked right out of the box. Did not use any backup-data this time, entered all my RX-settings manually. Had a short flight in level, horizon and acro mode - nop. Motor presets are correct now:

# Betaflight / KISSCC (KISS) 3.3.0 Feb 18 2018 / 15:02:34 (a5c9824ea) MSP API: 1.37
name -
resource BEEPER 1 B13
resource MOTOR 1 A08
resource MOTOR 2 B00
resource MOTOR 3 B14
resource MOTOR 4 B15

So diff is "clear" now of resource remappings:
# diff
# Betaflight / KISSCC (KISS) 3.3.0 Feb 18 2018 / 15:02:34 (a5c9824ea) MSP API: 1.37

feature TELEMETRY
map TAER1234
aux 0 0 0 1300 1700 0
aux 1 1 1 900 1250 0
aux 2 2 1 1300 1700 0
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set motor_pwm_protocol = DSHOT600
set current_meter = NONE
set battery_meter = ADC
profile 0
rateprofile 0

Thx all involved and have a nice sunday all.

PR #5217 (2018-02-19): Fixed motor order for KISSCC to match manual (QUADX1234).
Fixes #5201.
closed 2018-02-18 by Mode-1