betaflight/betaflight #5203: Craft name is overwriting some OSD settings in 3.3-RC1

si618, 2018-02-15:
After upgrading from 3.2.4 to 3.3-RC1 (also tested in build 611), vbat or mAh drawn are no longer showing in OSD if placed to right of craft name and on the same row.

OSD Layout: 
https://www.dropbox.com/s/y6n5o50wq9vg15z/vbat.png?dl=0

Disabling craft name enables display:
https://www.dropbox.com/s/ml2lgx94p7ncmzm/OSD2.jpg?dl=0

As does moving vbat from right of craft name: 
https://www.dropbox.com/s/4o22aeeuffeymio/OSD.jpg?dl=0

Interestingly, only "Main Batt Voltage" and "Mah Drawn" appear to be causing problems with craft name. Placing craft name to left of low voltage or warning settings works fine. Just guessing, but maybe it's related to the custom font used in those settings, rather than settings which just use plain text?

Confirmed craft name is set as two characters. Tried uploading different fonts, but had no effect. FC is ßFF4.

jflyper, 2018-02-15:
Current code treat the craft name as a 16 character fixed length item. Anything behind this region gets overwritten. Does this explain your issue?

si618, 2018-02-15:
Yes, it does partially explain it.

However, it only appears to be some settings whose text is being overwritten. For example, if I place craft name to close left of warnings or disarmed setting on same line, then they are **not** being overwritten, but to the left of vbat or mAh, then they are being overwritten. Not sure if there is some order of precedence involved? Sorry, I know nothing about the OSD code to be of any help.

Not sure about the wrapping either, as I tried with the craft name at the top of the window, and vbat was still hidden, with no wrapping onto the next line.

jflyper, 2018-02-16:
That is because display items are drawn sequentially. Something underneath the white space portion of the craft name will be overwritten. Something above the portion will not be overwritten; it overwrites the white space portion.

McGiverGim, 2018-02-16:
I suppose that writing spaces is the way to not erase the entire screen, and when value change, the element is totally refreshed.
Is there a way to change the craft name after powered up? If not is better to limit the length to the real craft name, 16 length is a lot of space wasted that can't be used by others elements. 
Other option is to put the craft name the first in the list of refreshing elements, in this way it will not overwrite others.

mikeller, 2018-02-17:
@McGiverGim:

> Is there a way to change the craft name after powered up? If not is better to limit the length to the real craft name, 16 length is a lot of space wasted that can't be used by others elements.

No there isn't, except maybe when connected to the configurator, and then I'd see it as acceptable to have a glitch in the display until reboot.

Do you have time to look into restricting the length of the string that is refreshed. (On the other hand, considering the above, not doing any refresh once the name was written should work, and would have the much desired effect of reducing the number of updated characters for a refresh when OSD over serial is used.

McGiverGim, 2018-02-17:
No I'm out. Maybe Monday or Tuesday.

JonnyPhenomenon, 2018-02-17:
ahhhh... that explains it.  I just rearranged my OSD after updating to the 2-12-18 firmware and noticed my RSSI display wouldn't show in the center of the top row with my craft name to the far left of it.  I explored a bit more and found that my "mah drawn" readout was also missing when it was _below_ the "avg cell voltage" indicator.  However, if I put "mah drawn" _above_ cell vltg, and pushed RSSI over to the right some, it all displays fine... 

Cliffnotes:
If I set up the OSD in the configurator as I have in the first frame, what I see in the goggles looks like the second frame.  If I rearrange it to look like the 3rd frame, then it displays correctly. 
[image]

anyway, I am glad I am not the only person to notice it.

mikeller, 2018-02-18:
#5222 should fix it. Can somebody please test?

si618, 2018-02-18:
Thanks guys!  Happy to test tonight.

JonnyPhenomenon, 2018-02-19:
I just flashed my BFF3 board with RC2, and the problem remains. - Am I to assume that RC2 does not have this update included?

mikeller, 2018-02-19:
@JonnyPhenomenon: That is correct - it wasn't ready in time for RC2, it will be in RC3. You can get a nightly build from http://ci.betaflight.tech/ to test it.

si618, 2018-02-19:
@JonnyPhenomenon Just missed it: https://github.com/betaflight/betaflight/compare/v3.3.0-RC2...master

Grab a build from here:
https://ci.betaflight.tech/job/Betaflight/lastBuild/

JonnyPhenomenon, 2018-02-19:
OK. just flashed it. the Craft name doesnt overwrite the RSSI valuu anymore. nicely done.

the Warnings field is still overwriting - at least some - fields.  if the "Mah drawn" field is to the right of the "warnings" field, "Mah drawn" will not display.
however, if the "Avg cell voltage" voltage is on the right of the warnings field, it shows up just fine.

mikeller, 2018-02-19:
Thanks for testing @JonnyPhenomenon. Reopening this, will have a look into the warnings field later on.

mikeller, 2018-02-22:
@JonnyPhenomenon: #5268 should fix the warnings issue as well. Can you test please?

PR #5222 (2018-02-18): Fixed problem with craft name overwriting elements in OSD.
Fixes #5203.
PR #5268 (2018-02-23): Fixed OSD warnings overlap to the right.
Fixes #5203.
closed 2018-02-23 by mikeller