betaflight/betaflight #5209: Betaflight 3.3 RC1 Beeper Problem on AirbotF4 (USB)

SchonRichtig, 2018-02-16:
Hi,

with BF 3.3RC1 the beeper won’t work anymore, when i disable the beeping on USB.
Yes, in the configurator this behavior is exactly what i want, but if i power it the quad with an lipo the beeper also stops working after init.
If I enable the beeper on usb in the configurator the beeper also works when the quad is powered up by a lipo.

etracer65, 2018-02-16:
I have 6 different quads all using AirbotF4 and all are working perfectly with BF 3.3RC1.  Double check your beeper settings in the CLI.  Mine look like this:

```
# version
# Betaflight / AIRBOTF4 (AIR4) 3.3.0 Feb 12 2018 / 13:04:13 (040c46da0) MSP API: 1.37

# beeper
Disabled:  ON_USB

```

SchonRichtig, 2018-02-16:
Here is a DIFF from my configuration.
Only disabled ON USB and the FC stops beeping on USB or powerd by lipo.

# diff

# version
# Betaflight / AIRBOTF4 (AIR4) 3.3.0 Feb 12 2018 / 13:04:13 (040c46da0) MSP API: 1.37

# name
name RealAccK130

# resources

# mixer

# servo

# servo mix

# feature
feature MOTOR_STOP
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -ON_USB

# map

# serial
serial 0 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 5 1700 2100 0
aux 1 1 0 900 1300 0
aux 2 2 0 1300 1700 0
aux 3 13 2 900 1300 0
aux 4 28 1 1700 2100 0
aux 5 35 4 1700 2100 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 6 s 1000

# master
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set gyro_filter_q = 450
set gyro_filter_r = 50
set rssi_channel = 8
set fpv_mix_degrees = 25
set serialrx_provider = SBUS
set blackbox_device = SPIFLASH
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 10
set failsafe_kill_switch = ON
set align_board_yaw = 90
set small_angle = 180
set deadband = 5
set yaw_deadband = 10
set pid_process_denom = 1
set debug_mode = NOTCH

# profile
profile 0

set dterm_lowpass_type = PT1
set setpoint_relax_ratio = 40
set dterm_setpoint_weight = 100
set crash_recovery = ON
set p_pitch = 61
set i_pitch = 55
set d_pitch = 20
set p_roll = 46
set i_roll = 45
set d_roll = 22
set p_yaw = 64
set i_yaw = 55

# rateprofile
rateprofile 0

set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 120

#

etracer65, 2018-02-16:
The only thing unusual I see is:

`rxfail 6 s 1000`

This is forcing your aux channel for the beeper into the range that makes it beep.  Try removing this and see what happens.

mikeller, 2018-02-18:
@SchonRichtig: Can you confirm if this has fixed it for you please?

jflyper, 2018-02-18:
Do CLI `defaults`, just in case.

mikeller, 2018-02-21:
Did this help, @SchonRichtig?

SchonRichtig, 2018-02-22:
Hi, 
sorry i have 2 little children and not much time :)
Yust loadet the defaults an configured the receiver.
Only configured beeper OFF on USB and the AUX3 to beeper.
The beeper won’t work powered by a lipo
____

# diff

# version
# Betaflight / AIRBOTF4 (AIR4) 3.3.0 Feb 18 2018 / 13:29:25 (f07bf40d1) MSP API: 1.37

# name

# resources

# mixer

# servo

# servo mix

# feature

# beeper
beeper -ON_USB

# map

# serial
serial 0 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 13 2 900 1300 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set serialrx_provider = SBUS

# profile
profile 0
# rateprofile
rateprofile 0
#

mikeller, 2018-02-22:
Could it be that your voltage detection isn't working properly, and the flight controller 'thinks' it's on USB all the time. Plug in a LiPo, and then check the 'Power State' frame on the 'Power & Battery' tab in configurator.

SchonRichtig, 2018-02-22:
@mikeller 
Oh man, you are right, that was the problem.
I use the camera osd to see the voltage because the airbot f4 hasn’t an osd.
After flashing to 3.3 I have forgotten to set the voltage settings.
Now it works.
But should the 3.3 RC2 not prevent arming if it (thinks) its on USB?
Thank you very much !!!
Kind regards

closed 2018-02-22 by SchonRichtig

mikeller, 2018-02-22:
@SchonRichtig: It does not prevent arming if the detected voltage stays at 0 - otherwise boards without voltage detection, or with voltage detection that is not connected would not work.