betaflight/betaflight #5211: debug_mode: for 3.3 changed from 8 to 6 for NOTCH in blackbox logs

Mechotronic, 2018-02-16:
I'm not sure if this is a planned change, but I noticed the debug_mode for the notch type changed from 8 (3.2.5) to 6 (3.3RC1)in the header for BB logs. I am unable to select the chart type DEBUG_NOTCH in BBE on 3.3RC1. 

There are also some parameter changes that break the header display panel. Some PID values are blank.

Mechotronic, 2018-02-17:
Looks like the enumeration is out of order now because the options are in a new order:

`#3.3RC1`
`debug_mode = NONE
Allowed values: NONE, CYCLETIME, BATTERY, GYRO, ACCELEROMETER, PIDLOOP, NOTCH, RC_INTERPOLATION, ANGLERATE, ESC_SENSOR, SCHEDULER, STACK, ESC_SENSOR_RPM, ESC_SENSOR_TMP, ALTITUDE, FFT, FFT_TIME, FFT_FREQ, RX_FRSKY_SPI, GYRO_RAW, MAX7456_SIGNAL, MAX7456_SPICLOCK, SBUS, FPORT, RANGEFINDER, RANGEFINDER_QUALITY, LIDAR_TF, CORE_TEMP, RUNAWAY_TAKEOFF`

`#3.2.5`
`debug_mode = NONE
Allowed values: NONE, CYCLETIME, BATTERY, GYRO, ACCELEROMETER, MIXER, AIRMODE, PIDLOOP, NOTCH, RC_INTERPOLATION, VELOCITY, DFILTER, ANGLERATE, ESC_SENSOR, SCHEDULER, STACK, ESC_SENSOR_RPM, ESC_SENSOR_TMP, ALTITUDE, FFT, FFT_TIME, FFT_FREQ, FRSKY_D_RX, GYRO_RAW`

mikeller, 2018-02-17:
@Mechotronic: This was caused by a cleanup in the firmware (to optimise flash space usage): #4910. Unfortunately, the corresponding update to the blackbox log viewer was forgotten.

Would you mind closing this issue, and opening it in https://github.com/betaflight/blackbox-log-viewer/issues? We'll update the log viewer when we get around to it.

closed 2018-02-18 by Mechotronic