betaflight/betaflight #5212: frsky fport test    High CPU load    ... flight shaky !!

adrianmiriuta, 2018-02-16:
# tasks
00 - (         SYSTEM)      9       8       1    0.5%    0.5%         0
01 - (            PID)   1865     118      54   22.5%   10.5%     18387
   - (           GYRO)   7462
02 - (            ACC)    988      28      18    3.2%    2.2%       791
03 - (       ATTITUDE)    498      29      20    1.9%    1.4%       434
04 - (             RX)  47619     128      12  610.0%   57.6%       868
05 - (         SERIAL)     99    3660      13   36.7%    0.6%       155
07 - (BATTERY_VOLTAGE)     49      13       5    0.5%    0.5%        10
08 - (BATTERY_CURRENT)     49      11       2    0.5%    0.5%         4
09 - ( BATTERY_ALERTS)      4      12       4    0.5%    0.5%         1
10 - (         BEEPER)     98      14       2    0.6%    0.5%        11
11 - (      TELEMETRY)    245      15       4    0.8%    0.5%        49
15 - (            CMS)     59      11       3    0.5%    0.5%         6
16 - (        VTXCTRL)      4      11       2    0.5%    0.5%         0
17 - (        CAMCTRL)      4       5       1    0.5%    0.5%         0
19 - (       PINIOBOX)     19       9       2    0.5%    0.5%         2
RX Check Function                  24       6                       433
Total (excluding SERIAL)                       643.0%   76.7%

# 

# diff
# Betaflight / LUX_RACE (LUX) 3.3.0 Feb 16 2018 / 19:20:10 (77559ad) MSP API: 1.37
name merica
resource MOTOR 2 C08
resource MOTOR 3 C07

feature AIRMODE
feature DYNAMIC_FILTER
map RETA1234
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
aux 0 0 1 1950 2075 0
aux 1 1 0 1525 1650 0
aux 2 2 0 1700 1825 0
aux 3 6 2 1950 2075 0
aux 4 7 3 1950 2075 0
aux 5 13 1 1425 1550 0
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set gyro_offset_yaw = -31
set min_check = 1000
set max_check = 2000
set rc_interp_ch = RPY
set serialrx_provider = FPORT
set serialrx_inverted = ON
set blackbox_device = NONE
set motor_pwm_protocol = DSHOT600
set current_meter = ADC
set ibata_scale = 265
set small_angle = 180
set deadband = 7
set yaw_deadband = 17
profile 0

set vbat_pid_gain = ON
set p_pitch = 61
set i_pitch = 37
set d_pitch = 31
set p_roll = 61
set i_roll = 37
set d_roll = 31
set p_yaw = 61
set i_yaw = 37
set d_level = 50
set horizon_tilt_effect = 50
set horizon_tilt_expert_mode = ON
rateprofile 0

set thr_mid = 14
set thr_expo = 67
set tpa_rate = 70

#

teshko, 2018-02-20:
I have this same issue on F4FLAME and SPRACINGF4EVO which don't even arm. On OMNIBUSF4SD is not so bad, but lua scripts don't work right. I'm using uninverted XSR-M.  Tested both on RC1 and RC2.

```
# diff

# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.3.0 Feb 18 2018 / 13:55:02 (f07bf40d1) MSP API: 1.37

# name

# resources
resource MOTOR 6 NONE
resource CAMERA_CONTROL 1 A08

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 2100 0
aux 1 1 1 1700 2100 0
aux 2 13 2 1300 2100 0
aux 3 28 0 1700 2100 0
aux 4 35 2 1700 2100 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 6 s 1000
rxfail 7 s 1000

# master
set min_check = 1010
set max_check = 1990
set rssi_channel = 8
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set dshot_idle_value = 300
set motor_pwm_protocol = DSHOT600
set small_angle = 179
set pid_process_denom = 1
set tlm_inverted = ON
set vtx_channel = 4
set vtx_power = 2
set vtx_freq = 5800

# profile
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 125
set pitch_rc_rate = 125
set yaw_rc_rate = 105
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 65
```

```
# tasks
Task list             rate/hz  max/us  avg/us maxload avgload     total/ms
00 - (         SYSTEM)      9       3       0    0.5%    0.0%         1
01 - (       GYRO/PID)   7812      56      43   44.2%   34.0%     88647
02 - (            ACC)    978      19      10    2.3%    1.4%      2557
03 - (       ATTITUDE)     99      18      10    0.6%    0.5%       254
04 - (             RX)  17857      45       2   80.8%    4.0%      2818
05 - (         SERIAL)     99    2852       1   28.7%    0.5%       244
07 - (BATTERY_VOLTAGE)     49       6       2    0.5%    0.5%        23
08 - (BATTERY_CURRENT)     49       4       0    0.5%    0.0%         7
09 - ( BATTERY_ALERTS)      4       3       1    0.5%    0.5%         1
10 - (         BEEPER)     99       9       1    0.5%    0.5%        27
13 - (           BARO)     43      21       8    0.5%    0.5%        86
15 - (       ALTITUDE)     39      26      19    0.6%    0.5%       197
17 - (      TELEMETRY)    248       9       2    0.7%    0.5%       152
20 - (            OSD)     59     504      29    3.4%    0.6%       471
22 - (            CMS)     59       4       1    0.5%    0.5%        12
23 - (        VTXCTRL)      4      17       1    0.5%    0.5%        11
24 - (        CAMCTRL)      4       2       1    0.5%    0.5%         0
26 - (    ADCINTERNAL)      0       4       2    0.5%    0.5%         0
27 - (       PINIOBOX)     19       6       1    0.5%    0.5%         4
RX Check Function                  13       2                      2228
Total (excluding SERIAL)                       138.1%   46.0%
```

```
# diff

# version
# Betaflight / SPRACINGF4EVO (SP4E) 3.3.0 Feb 18 2018 / 14:02:31 (f07bf40d1) MSP API: 1.37

# name

# resources

# mixer

# servo

# servo mix

# feature
feature -RSSI_ADC
feature -LED_STRIP
feature -TRANSPONDER

# beeper

# map
map TAER1234

# serial

# led

# color

# mode_color

# aux
aux 0 0 0 1300 2100 0
aux 1 1 1 1700 2100 0
aux 2 13 2 1300 2100 0
aux 3 19 1 1300 1675 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set mag_hardware = NONE
set rssi_channel = 8
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set motor_pwm_protocol = DSHOT600
set current_meter = ADC
set ibata_scale = 235
set osd_vbat_pos = 2103
set osd_rssi_pos = 2082
set osd_tim_1_pos = 2487
set osd_tim_2_pos = 2519
set osd_throttle_pos = 2466
set osd_current_pos = 2498

# profile
profile 0

# rateprofile
rateprofile 0
```

```
# tasks
Task list             rate/hz  max/us  avg/us maxload avgload     total/ms
00 - (         SYSTEM)      4       4       0    0.5%    0.0%         0
01 - (            PID)   1851      45      17    8.8%    3.6%     16404
   - (           GYRO)   7407
02 - (            ACC)    499      15       9    1.2%    0.9%       790
03 - (       ATTITUDE)     49      16      10    0.5%    0.5%        84
04 - (             RX) 125000      45       1  563.0%   13.0%      8707
05 - (         SERIAL)     16    2646       1    4.7%    0.5%        41
07 - (BATTERY_VOLTAGE)     25       5       1    0.5%    0.5%         8
08 - (BATTERY_CURRENT)     24       4       1    0.5%    0.5%         2
09 - ( BATTERY_ALERTS)      2       5       1    0.5%    0.5%         0
10 - (         BEEPER)     16      10       0    0.5%    0.0%         7
13 - (           BARO)      7     115      75    0.5%    0.5%       235
14 - (       ALTITUDE)      6      27      21    0.5%    0.5%        58
16 - (      TELEMETRY)     41       8       2    0.5%    0.5%        33
19 - (            OSD)     10     128       7    0.6%    0.5%        31
21 - (            CMS)      9       5       1    0.5%    0.5%         3
22 - (        VTXCTRL)      0     130       0    0.5%    0.0%         0
23 - (        CAMCTRL)      0       3       0    0.5%    0.0%         0
25 - (    ADCINTERNAL)      0       5       2    0.5%    0.5%         0
26 - (       PINIOBOX)      0       4       1    0.5%    0.5%         1
RX Check Function                  11       2                     12301
Total (excluding SERIAL)                       580.6%   23.0%
```

mikeller, 2018-02-20:
@adrianmiriuta: What RX model are you using?

adrianmiriuta, 2018-02-20:
@mikeller 
R-XSR_FPORT_LBT180111

teshko, 2018-02-20:
Guys I can probably debug/test this one on my own, just give me some setup pointers since I am unfamiliar with the platform. I am an EE and I have experience debugging embedded systems, and I also own STLink V2 and oscilloscope. I've built and flashed custom configured OBF4 last night (that works), and I prefer to use Linux for these kind of build environments. I will not be able to do any work before Thursday afternoon (2018-02-22) though.

mikeller, 2018-02-22:
@adrianmiriuta, @teshko: Can you please test #5269?

adrianmiriuta, 2018-02-22:
@mikeller 
Indoor test.
seems ok.

# tasks
00 - (         SYSTEM)      9       8       2    0.5%    0.5%         0
01 - (            PID)   1923     118      53   23.1%   10.6%      4426
   - (           GYRO)   7692
02 - (            ACC)    917      32      19    3.4%    2.2%       231
03 - (       ATTITUDE)    496      30      20    1.9%    1.4%       120
04 - (             RX)    652     120      62    8.3%    4.5%       181
05 - (         SERIAL)     99    3523       9   35.3%    0.5%        27
07 - (BATTERY_VOLTAGE)     49      14       6    0.5%    0.5%         3
08 - (BATTERY_CURRENT)     49      10       2    0.5%    0.5%         1
09 - ( BATTERY_ALERTS)      4      13       4    0.5%    0.5%         0
10 - (         BEEPER)     99      16       2    0.6%    0.5%         3
11 - (      TELEMETRY)    249      15       4    0.8%    0.5%        15
15 - (            CMS)     59      11       3    0.5%    0.5%         1
16 - (        VTXCTRL)      4      11       2    0.5%    0.5%         0
17 - (        CAMCTRL)      4       6       2    0.5%    0.5%         0
19 - (       PINIOBOX)     19      12       3    0.5%    0.5%         0
RX Check Function                  23      10                        29
Total (excluding SERIAL)                        42.1%   23.7%

#

# diff
# Betaflight / LUX_RACE (LUX) 3.3.0 Feb 22 2018 / 18:19:11 (bd4c970) MSP API: 1.37
name merica
resource MOTOR 2 C08
resource MOTOR 3 C07

feature AIRMODE
feature DYNAMIC_FILTER
map RETA1234
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
aux 0 0 1 1950 2075 0
aux 1 1 0 1525 1650 0
aux 2 2 0 1700 1825 0
aux 3 6 2 1950 2075 0
aux 4 7 3 1950 2075 0
aux 5 13 1 1425 1550 0
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set gyro_offset_yaw = -31
set min_check = 1000
set max_check = 2000
set rc_interp_ch = RPY
set serialrx_provider = FPORT
set serialrx_inverted = ON
set blackbox_device = NONE
set motor_pwm_protocol = DSHOT600
set current_meter = ADC
set ibata_scale = 265
set small_angle = 180
set deadband = 7
set yaw_deadband = 17
profile 0

set vbat_pid_gain = ON
set p_pitch = 61
set i_pitch = 37
set d_pitch = 31
set p_roll = 61
set i_roll = 37
set d_roll = 31
set p_yaw = 61
set i_yaw = 37
set d_level = 50
set horizon_tilt_effect = 50
set horizon_tilt_expert_mode = ON
rateprofile 0

set thr_mid = 14
set thr_expo = 67
set tpa_rate = 70

#

teshko, 2018-02-22:
@mikeller:
OMNIBUSF4SD seems absolutely fine, I even took it for a flight and had no issues. Telemetry and scripts are working fine.
Indoor test of F4FLAME (OBF4) also worked, but with no telemetry. I decided to swap receivers, and telemetry started working fine. It appears that two of three receivers are **"acting up"** and previously caused 100% load. I will also test SPRACINGF4EVO later tonight, but I'm pretty sure it will be fine.

teshko, 2018-02-22:
@mikeller:
Confirmed correct operation on SPRACINGF4EVO. I definitely have two buggy receivers which exacerbated the issue.

PR #5269 (2018-02-23): Fixed efficiency of FPort protocol.
Fixes #5212.
closed 2018-02-23 by mikeller