betaflight/betaflight #5214: Biquad RC+FIR2: Potential implementation or analysis problem

fujin, 2018-02-16:
User has reported the inability to fly with the Biquad RC+FIR2 while the FKF works for them: https://github.com/betaflight/betaflight/pull/4965#issuecomment-366170285 / https://www.rcgroups.com/forums/showpost.php?p=39181579&postcount=53797

rs2k posted https://www.rcgroups.com/forums/showpost.php?p=39184983&postcount=727 here some logs for 1000 runs from a filter test program showing a difference as well.

It is entirely possible that I have made a terrible implementation error or some factor in the analysis was not completely accurate, so I would like to rule out both.

etracer65, 2018-02-16:
The user posted an update and indicated that the differences in the dterm lowpass were the cause.

https://www.rcgroups.com/forums/showpost.php?p=39186664&postcount=53818

fujin, 2018-02-16:
Thank goodness for that, @etracer65. Thank you.

closed 2018-02-16 by fujin