betaflight/betaflight #5218: Issue with "3D on a switch", airmode or other mechanic doesnt work properly. 3.3.0 RC1

thongraegu, 2018-02-17:
I have been flying this "3D on a switch" mode as long as its been there and on recent RC1 release there is new problem. Quadcopter has a lot of wobble when on zero throttle like falling down after punch outs. This wobble wasnt there on earlier builds like #430. It feels like airmode doesnt work properly and copter has movement when falling down with no throttle and not touching sticks.

Small video where I let go off sticks after a few punch outs and you can see this wobble:
https://youtu.be/ZjzmLvqoIZw

# feature
feature TELEMETRY
feature 3D
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper

# map
map TAER1234

# serial
serial 5 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1675 2075 0
aux 1 13 1 1300 1700 0
aux 2 35 1 1750 2100 0
aux 3 38 2 1300 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set gyro_filter_q = 300
set gyro_filter_r = 80
set gyro_stage2_filter_type = FKF
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1020
set max_check = 2000
set rssi_channel = 8
set rc_interp = MANUAL
set rc_interp_int = 14
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT1200
set vbat_min_cell_voltage = 30
set vbat_warning_cell_voltage = 30
set ibata_scale = 460
set beeper_dshot_beacon_tone = 4
set yaw_motors_reversed = ON
set pid_process_denom = 1
set osd_warnings = 63
set osd_vbat_pos = 2370
set osd_rssi_pos = 2061
set osd_tim_2_pos = 2422
set osd_mah_drawn_pos = 2401
set osd_warnings_pos = 2281
set vcd_video_system = 2

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_notch_hz = 0
set anti_gravity_gain = 7000
set dterm_setpoint_weight = 100
set p_pitch = 63
set i_pitch = 65
set p_roll = 53
set i_roll = 60
set p_yaw = 90
set i_yaw = 60

# rateprofile
rateprofile 0

set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80
set tpa_rate = 15

mikeller, 2018-02-17:
@thongraegu: Can you please provide the changeset id of your build (part of the first line of the `diff` / `dump`)?

mikeller, 2018-02-17:
@thongraegu:

> set gyro_stage2_filter_type = FKF

This is not something that is implemented in any official build. Can you please switch to an official build and try to replicate, and if it replicates, open a new issue.

closed 2018-02-17 by mikeller