betaflight/betaflight #5220: CPU load percentage in OSD

JonnyPhenomenon, 2018-02-17:
Could the CPU load percentage be added as an OSD item?

It might be useful for people who want to explore the various filters and loop frequencies to be able to monitor the CPU load in flight....

mikeller, 2018-02-17:
Duplicate of #4934.

closed 2018-02-17 by mikeller

JonnyPhenomenon, 2018-02-17:
@mikeller  - uh, I just looked at 4934, and it doesnt look like its related to my OSD feature request... I did do a search for CPU/OSD related feature requests before I posted it... am I missing something?

mikeller, 2018-02-17:
@JonnyPhenomenon: Sorry, that was me jumbling numbers. The issue this duplicates is #4394.

JonnyPhenomenon, 2018-02-18:
ahh yes. I thought that might be the case.  I tried looking through the issue history for 4934, and 4943, but did not go back to the 4300's. - I swear, I really did search for OSD and CPU in the same search, but came up with nothing...  

Anyway, the issue is closed, according to #4394:
> wind0r commented on Oct 19, 2017
> Since Betaflight 3.1 CPU usage only is calculated when in CLI.
> This saves a lot of cpu cycles but on the "downside" the OSD element got removed.
> Which isnt even that usefull ;) Just check the CPU Usage in CLI. Real usage will be below that value. 

I can live with that. :)
thanks!