betaflight/betaflight #5225: [Suggested Feature] Resource Remapping Reset

rhys100, 2018-02-18:
Is it possible to add a command to reset the resource list to default?

There is times where I have remapped something to NONE without taking note what the reference was and then im stuck and have to reset everything.

A simple command such as resource reset could do the trick and add to the wiki?

https://github.com/betaflight/betaflight/wiki/Betaflight-resource-remapping

McGiverGim, 2018-02-18:
The `diff all defaults` shows the actual and original value for all the changes, I don't know if for resource too. If not maybe can be added to this command.

rhys100, 2018-02-18:
ok didnt know that command existed, to fix before i typed diff, then reset fc and pasted them back in without the resource remap from diff.

a simple resource reset would be good tho ;)

mikeller, 2018-02-18:
@rhys100: I don't think it makes much sense to elevate the resource configuration above all others for this purpose. The same feature could be requested for other feature groups like modes, PIDs, LED, you name it, leading to a lot of flash space being allocated to something that can easily be done off-board, by running a `diff` or `dump`, and then editing the result.

Another point in this is that for a lot of other feature groups, a 'clean all' is implemented in the configurator (cf. PIDs, LED). Obviously, since configurator support for resource management has not been implemented yet, this isn't possible for resources. But resource management in configurator is still on our to-do list, and once it's done, resetting of all resources in configurator is definitely something we'll look into.

For now I suggest we close this.