betaflight/betaflight #5227: Smart Audio Pitmode on/off question

AndersHoglund, 2018-02-18:
In src/main/io/vtx_smartaudio.c, line 814, you can find this piece of code:
```
    if (onoff) {
        // SmartAudio can not turn pit mode on by software.
        return;
    }
```

Is there anyone who knows the background and reason for this limitation? I have seen comments on RCG that VTX Pitmode on/of does not work. As i do not have an SA VTX, I can not test this. I only have IRC Tramp HV, and no problems switching Pitmode on and off.

dcanham, 2018-02-20:
@AndersHoglund I believe I've read elsewhere that since the VTX's have to be restarted in order to enter pit mode, they can't be done via SA because they don't have a bootstrap in order to reboot the device. Please correct me if I'm wrong though.

codecae, 2018-02-20:
Sadly, the hardware will not respond to enabling pit mode after power-up.  I've asked TBS if this could be implemented in future versions.

AndersHoglund, 2018-02-20:
OK, thnx for the info. Too bad.

closed 2018-02-20 by AndersHoglund