betaflight/betaflight #5232: Betaflight developers should stop acting like stupid teenagers.

respet, 2018-02-19:
Locking a discussions because you are in the wrong is a shameful move and your users are watching. You only care for being right. You have no honor of truth. It is uncalled for and you are all acting like school children. You should be ashamed of yourselves. This is not how open development should be. You do not choose to accept or deny code based on whether or not you like someone. You do it for the good of the people. This is high school drama and will only lead to fracturing the community. GROW UP! 

https://github.com/betaflight/betaflight/pull/5231
https://github.com/betaflight/betaflight/pull/5215
https://github.com/betaflight/betaflight/pull/4965

@fujin 
@mikeller 
@apocolipse
@rs2k
@orneryd

Bradw-17, 2018-02-19:
They locked that convo because they are busy trying to develop their project, and someone was wasting time creating childish posts, and then whining even further on this post.

This is hypocritical nonsense and a distraction which has no place in software development.

mikeller, 2018-02-19:
Closing as off-topic.

closed 2018-02-19 by mikeller