betaflight/betaflight #5233: Quads Don't ARM - NOGYRO error

ALopes90, 2018-02-19:
Hi, I've spend a lot of time searching for someone with the same problem as me and I can't find it so I think is time to share it in here, because or it is a firmware problem or I'm the unluckiest man in FPV right now.

In the past mouth with me preparing for this year competitions I started updating my ESC and Flight Controllers, I had a competition earlier and I saw what I was missing from not have Turtle mode, so I updated and configure the 4 Quads I'm using at the moment, didn't even used dump for a clean start.

Right now after some training and testing with almost no crashes i only have a quad that works, the other 3 don't Arm saying in the osd NOGYRO and when plugged into the computer the gyro symbol in betaflight is off.

Already tried changing to old firmwares and to the new 3.3 and the problem stays the same, already tried stick commands to make the gyro calibrate and nothing, the quads don't make the last 2 beeps.

The first that was a Omnibus F4 micro 20x20 (probably a banggood clone), I just thought that was a crappy controller, them it was a HGLRC F4 micro 20x20 and I thought that was me and tried every thing i could remember and last week it was one almost new Betaflight F3 original board from FPVModel that one day when i was going to test a set of props just stop arming.

After talking to my friends i realize that many of them had the same problem with controllers like the  Omnibus F4 micro 20x20 from banggood and others originals that after a little crash just stopped working.

I'm just posting here because the only thing that all of these cases have in common when it happen was some Betaflight version with the Dynamic Filter, of course i'm not blagging it, I just think that this needs to be analyzed for you guys that in the case that there is a problem in there it can be fixed.

Thank you for your time, keep up the good work you all are doing.
Alexandre Lopes

etracer65, 2018-02-19:
Try going into the CLI and type `defaults`.  The flight controller will reboot.  Then connect with the configurator to see if the gyro is recognized.

If not then it's likely you have a hardware failure on the board.  Alternately you may have flashed the wrong target.

ALopes90, 2018-02-19:
I done it on the Betaflight F3 flight controller and it stays the same.

I think that this is a hardware problem but I can't accept that is a random problem, I'm suggesting here it that something is causing this.

etracer65, 2018-02-19:
Something probably is - most likely crashes.  If you're suggesting that somehow the firmware is damaging the gyro - then no, that's impossible.

Try going back to an old version of Betaflight like 2.9 or even older.  If the gyro is not detected then you have your answer.

An old trick is to reflow your board using a hot-plate of oven.  It's not uncommon for the gyro chip to work loose after significant impacts.  On all my builds I encapsulate the gyro chip in epoxy to minimize the chances of this occurring.

mitsinsar, 2018-02-19:
I have the same problem. I flew once. After an crash which is not damaging, i flew again. Came back home. The day after, went out to fly again. I flew with one battery. When I change it, it said "NOGYRO". came back home. the day after worked very well. no errors etc. I flew like 10-15 mins. But now i have the same problem again. I've read something about gyro. If you make your PID loop frq too high, CPU ignores the gyro for PID frq. Try to make low your PID frequency. That did not work on me, but maybe it works on yours.

mikeller, 2018-02-19:
@mitsinsar: This sounds very much like a broken solder joint - try the method described in https://github.com/betaflight/betaflight/issues/5233#issuecomment-366758456.

JonnyPhenomenon, 2018-02-19:
INTERESTING!  That NOGYRO error is exactly what grounded my first BFF3 board, which was Also a genuine FPVmodel first gen...  I had banged it around pretty hard, so I just assumed it was done for and replaced it with a new one.    the USB port on the old one was stove up pretty bad so it made reflashing it with experimental firmwares quite tricky..  I know the default PID loop and gyro loop frequency (for the BFF3 board at least) is 8k/2k.   and I didnt change the setting... Now, I see Joshua Bardwell suggests to use 4k/4k with dynamic filtering enabled...    

that being said, I _am_ inclined to agree with @mikeller on this, at least to some degree.  these components we use are not using leaded solder on them, and they are unfortunately much more susceptible to contact deterioration due to vibration or impact...  So the law of parsimony says "the simplest solution is probably the right one".  The issue is _probably_ the hardware.....

mitsinsar, 2018-02-20:
@mikeller Probably you are right. Because my gyro chip touches to VTX cover which is really 
 hot. I will send mail to the seller, if they dont do anything, i will resolder and comment here about the result

mitsinsar, 2018-02-21:
UPDATE:
I resolder my gyro's legs. Everything works fine.

ALopes90, 2018-02-21:
@mitsinsar how do you have resolder the gyro? in the oven?

mitsinsar, 2018-02-21:
@ALopes90  no, I just used my soldering iron. I put some solder on it, and swipe it through the legs. I already gave up, i was going to buy new FC. So I tried this as my last chance and it worked pretty well. Btw, I've changed the sponge under gyro chip. I put 2 sided tape which is shorter than sponge. Now, it doesnt touch to VTX's cover. (I have furibee x215 pro)

closed 2018-02-21 by mikeller