betaflight/betaflight #5234: Sonar with Matek F722

diver3416, 2018-02-19:
Hi, 

I try to use a sonar with my FC MATEK F722.
I tried to modify code target.h and target.mk (from betaflight on this repository) like this:

For target.h, I added that

*********** RangeFinder ************* ///////// ADDED
#define USE_RANGEFINDER
#define USE_RANGEFINDER_HCSR04
#define RANGEFINDER_HCSR04_TRIGGER_PIN       PB1
#define RANGEFINDER_HCSR04_ECHO_PIN          PA8

For target.mk, I added that
drivers/rangefinder/rangefinder_hcsr04.c

When I launch compilation (make TARGET=MATEKF722), I have that error message:

./src/main/drivers/rangefinder/rangefinder_hcsr04.c: In function 'hcsr04Detect':
./src/main/drivers/rangefinder/rangefinder_hcsr04.c:203:153: error: 'EXTI_Trigger_Rising_Falling' undeclared (first use in this function)
         EXTIConfig(echoIO, &hcsr04_extiCallbackRec, NVIC_PRIO_SONAR_EXTI, EXTI_Trigger_Rising_Falling); // TODO - priority!
                                                                                                                                                         ^                       ./src/main/drivers/rangefinder/rangefinder_hcsr04.c:203:153: note: each undeclared identifier is reported only once for each function it appears in

Can you explain me ?

Does it work with other flight controller (for NAZE i see in target.h that rangefinder code is commented) ?

Thanks in advance for your help

David

jflyper, 2018-02-20:
F7 does not support HCSR04, or, does not support `EXTIConfig` of this calling prototype.

To fix it, `EXTIConfig` must be reviewed and modified to accept edge polarity.

diver3416, 2018-02-20:
Thanks for this answer.

closed 2018-02-20 by diver3416

ubanov, 2019-09-07:
Hi,

  Today I have tried to make what you described above. And with current version (4.0), and 1 year and a half before yours... I have compiled firmware for MatekF722 with sonar support (not tested in real FC to see if it really works).

(in compilation window it appears that file rangefinder_hcsr04.c is compiled)
[...]
%% (size optimised) ./src/main/drivers/compass/compass_lis3mdl.c
**%% (optimised) ./src/main/drivers/rangefinder/rangefinder_hcsr04.c**
%% (speed optimised) ./src/main/drivers/max7456.c
[...]

some day I will connect the sensor in my quad and make real test.

Regards