betaflight/betaflight #5240: No response from the bootloader, programming: FAILED

mhr63, 2018-02-20:
I bought a king kong tiny 7 I installed the following receiver on it

https://hobbyking.com/en_us/orangerx-r614xn-dsm2-dsmx-nano-indoor-diy-receiver-with-cppm-pwm-spektrum-jr-multirotor-quadcopter.html

Wiring method :

vcc - > +5v
gnd -> gnd
cppm -> Rx
I made all the settings from inside the betaflight app
I bind the receiver with the radio (dx9)
But the motors do not work
I connected back to the program that I change my settings
In the port section I was changing options That betaflight freezed
I removed the cable from the FC And when I plugged in again, it did not get any more
I told myself that will flash FC But it does not flash And the following error

No response from the bootloader, programming: FAILED

What to do now?

mhr63, 2018-02-20:
ok
recovery and connect to betafkifgt
but untile But I still have trouble getting the engines off
in reciver section Nothing works

Arakon, 2018-02-20:
This is not a code issue. Please ask questions about setup etc on forums such as rcgroups.

closed 2018-02-20 by jflyper