betaflight/betaflight #5244: I have to rotate the board by -90 degrees on yaw after updating Firmware to 3.2.4 / 3.3

kolossboss, 2018-02-20:
Hi,
I have a strange issue.
A few weeks ago I updated my little 3inch quad from 3.2.x to 3.2.4.(**DYS AIO F4 with OmnibusF4SD**) 
I always do a full chip erase and copy my diff back in the cli.
After that I made a little test run and my copter is totally flipping out. 
So I went back to Betaflight configurator and saw that the orientation of the board is wrong. I have no idea why. The little arrow on the board is shows straight ahead and I flow with 0 degrees on yaw for month now.
Anyway, I changed the orientation an yaw to -90 degrees and everything works as expected and I did not think about it any further.

TODAY I updated my Racerstar StarF4S (F4 with OmnibusF4) from BF 3.2.3 to BF 3.3. 
Again, full chip erase and copy diff to CLI.
Again a little testflight and again, to copter is flipping out.
I went back to Betaflight Configuration and saw the same strange behavoir. The Orientation was wrong and I had to change YAW to -90 degrees. The Racestar is also build in in the right direction (straight ahead)

So what the hell is going on? Do something completely wrong? Or is that a problem with newer versions of BF?
I wonder that nobody had this issue before.

Thx a lot

etracer65, 2018-02-20:
Can you provide the configuration you pasted back into the CLI after upgrading?  Also what Configurator version?

kolossboss, 2018-02-20:
Hi, thx for reply.

I tried a bit more and I flashed bf again 3.3 to the board with Full Chip erase and without the diff. Its the same issue. The orientation is wrong. 

I'm using the stand alone Betaflight Configurator with Version 10.1.0

Thats the diff for the StarF4S Board.

`Entering CLI Mode, type 'exit' to return, or 'help'

# version
# Betaflight / OMNIBUSF4 (OBF4) 3.2.3 Dec 11 2017 / 07:56:31 (cb962eda1) MSP API: 1.36

# diff

# version
# Betaflight / OMNIBUSF4 (OBF4) 3.2.3 Dec 11 2017 / 07:56:31 (cb962eda1) MSP API: 1.36

# name

# resources

# mixer

# servo

# servo mix

# feature
feature MOTOR_STOP
feature LED_STRIP
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper

# map

# serial
serial 0 64 115200 57600 0 115200
serial 5 2048 115200 57600 0 115200

# led
led 0 0,15::ATI:0
led 1 1,15::ATI:0
led 2 2,15::ATI:0
led 3 3,15::ATI:0
led 4 4,15::ATI:0
led 5 5,15::ATI:0

# color

# mode_color

# aux
aux 0 0 0 1700 2100
aux 1 1 1 900 1125
aux 2 2 1 1300 1700
aux 3 13 2 1850 2100

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set acc_trim_pitch = 2
set acc_trim_roll = -14
set baro_hardware = NONE
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT600
set gps_provider = UBLOX
set pid_process_denom = 1
set osd_vbat_pos = 2466
set osd_rssi_pos = 40
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2498
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 2486
set osd_mah_drawn_pos = 353
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_notch_hz = 0
set p_pitch = 29
set i_pitch = 27
set d_pitch = 20
set p_roll = 34
set i_roll = 33
set d_roll = 23
set p_yaw = 46
set i_yaw = 37

# rateprofile
rateprofile 0

set rc_rate = 125
set rc_rate_yaw = 110
set tpa_rate = 20

# `

etracer65, 2018-02-20:
I don't see anything obvious in your diff.

Don't think you should have been using the OMNIBUSF4SD target.  All the documentation I can find for the DYS board specifies the OMNIBUSF4 target.

There was a new DYSF4PRO target added in BF3.2 as a variant of OMNIBUSF4 that's probably what you should be using rather than OMNIBUSF4.

On the Racerstar StarF4S I can't find any documentation that explicitly says which target to use.  It has a strange layout so that when installed correctly the USB port will come out the back.

I haven't seen any other reports like yours so I'm not sure what's going on other than you yave the boards installed in the wrong orientation - but you said that's not the case.

Can you try flashing back to your original 3.2.0 configuration to see it the alignment is still incorrect?

kolossboss, 2018-02-20:
Ok I can try to flash the DYS4PRO target. And I can try OMNIBUSF4 but I did not changed the target and the OMNIBUSF4 and OMNIBUSF4SD are mostly the same.

Yes the USB port comes out the back an the StarF4S. I can make a photo that both of the board are correct installed. I flow them a few month now without issues :D 

I will try to flash the original fw. Very strange.

kolossboss, 2018-02-21:
Thanks for helping

I think choosed the wrong target. :-( 👎 
My Test resultes with the DYS AIO F4 OmnibusF4:

OmnibusF4SD v3.3 Target = Wrong orientation
OmnibusF4SD v3.2.0 Target = Wrong orientation
OmnibusF4 v3.2.0 Target = right orientation
OmnibusF4 v3.3.0 Target = right orientation
DYSF4PRO v.3.3 Target = right orientation

My Test resultes with the StarF4S OmnibusF4:
OmnibusF4SD v3.3 Target = Wrong orientation
OmnibusF4SD v3.2.0 Target = Wrong orientation
OmnibusF4 v3.2.0 Target = right orientation
OmnibusF4 v3.3.0 Target = right orientation

closed 2018-02-21 by kolossboss