betaflight/betaflight #5245: Lidar with F7 Board (MATEK F722)

diver3416, 2018-02-20:
Hello,
I would like to know if I can use lidar (TF mini) as rangefinder with my FC Matek F722 (on uart port) without modification of code
If simple modification of code in target.h (.c/.mk) are required, I would appreciate your help.
Thanks
David

diver3416, 2018-02-20:
If it can work, what do I have to do in serial ports of betaflight configurator. 
Thanks

etracer65, 2018-02-20:
@jflyper Can you answer this one?

jflyper, 2018-02-21:
This should work.

1. Configure it in target.h
```
#define USE_RANGEFINDER
#define USE_RANGEFINDER_TF
```

2. CLI
```
serial <n> 32768 115200 57600 0 115200
set rangefinder_hardware = TF02
```

diver3416, 2018-02-21:
Thank you very much. 
I will try that. 
David

diver3416, 2018-02-21:
For TF mini is it
set rangefinder_hardware = TFMINI 
???

jflyper, 2018-02-21:
Ah, yes, of course ;)

diver3416, 2018-02-21:
Hello jflyper,
First thank you for your help
I tried to make compilation on mac and windows, but I have some errors in rangefinder.c.
(compilation goes well until errors in rangefinder.c)
See below:

./src/main/sensors/rangefinder.c:2:9: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'file'
  * This file is part of Cleanflight.
         ^~~~
./src/main/sensors/rangefinder.c:2:9: error: unknown type name 'file'
./src/main/sensors/rangefinder.c:10:30: error: unknown type name 'without'
  * but WITHOUT ANY WARRANTY; without even the implied warranty of
                              ^~~~~~~
./src/main/sensors/rangefinder.c:10:43: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'the'
  * but WITHOUT ANY WARRANTY; without even the implied warranty of
                                           ^~~
./src/main/sensors/rangefinder.c:10:43: error: unknown type name 'the'
In file included from c:\dev\betaflight\tools\gcc-arm-none-eabi-7-2017-q4-major\arm-none-eabi\include\stdint.h:14:0,
                 from c:\dev\betaflight\tools\gcc-arm-none-eabi-7-2017-q4-major\lib\gcc\arm-none-eabi\7.2.1\include\stdint.h:9,
                 from ./src/main/sensors/rangefinder.c:19:
c:\dev\betaflight\tools\gcc-arm-none-eabi-7-2017-q4-major\arm-none-eabi\include\sys\_stdint.h:20:9: error: unknown type name '__int8_t'
 typedef __int8_t int8_t ;

jflyper, 2018-02-21:
Your rangefinder.c is broken.

diver3416, 2018-02-21:
oh yeah I had uncommented the beginning of the file by mistake.
Yeah compilation is successful.
THANK YOU !!!!!!!!