betaflight/betaflight #5248: OSD text "CRASH FLIP" follows TX switch, not the actual state of the craft

Mechotronic, 2018-02-20:
1. Flip the crash flip switch on your TX
2. Flip pre-arm and then arm. "CRASH FLIP" will display on the OSD.
3. Flip the crash flip switch on the TX to off. "CRASH FLIP" will disappear.
4. move the RP stick and the craft is still in crash flip mode.

I feel like this particular display should follow the mode state as it tends to be a high stress situation when you use crash flip. Motor health is a concern too. Either that or disable crash flip mode immediately when the switch is deactivated.

etracer65, 2018-02-21:
@Mechotronic I think you're absolutely correct - the OSD warning should match the internal state of crash-flip (ie. how the quad is going to behave).  I've submitted https://github.com/betaflight/betaflight/pull/5253 to fix.

However I don't have a quad with an OSD at the moment so if you're willing to help test let me know what target you need and I'll build you a hex file to try.

Mechotronic, 2018-02-21:
Sure. I have a CLRacing F4 to test on and an omnibus F3. I can test in about 8 hours after work. Thanks for working on this.

etracer65, 2018-02-21:
@Mechotronic Here are the hex files for both.  No need to test both - just whichever is more convenient.

[betaflight_3.3.0_CLRACINGF4.hex.zip](https://github.com/betaflight/betaflight/files/1744188/betaflight_3.3.0_CLRACINGF4.hex.zip)
[betaflight_3.3.0_OMNIBUS.hex.zip](https://github.com/betaflight/betaflight/files/1744190/betaflight_3.3.0_OMNIBUS.hex.zip)

Mechotronic, 2018-02-21:
Works perfectly. Well done. I ran it through a bunch of scenarios and the display always followed the state.

etracer65, 2018-02-21:
Thanks for the testing and feedback.

closed 2018-02-22 by mikeller