betaflight/betaflight #5251: VTX low power on disarm

Gaz-B, 2018-02-21:
This feature doesn't work on the later builds of 3.3

AndersHoglund, 2018-02-21:
Seems to work fine with Spektrum Tx/Rx and Tramp HV. N.b. have not measured actual VTX power outout, but it reports 200mW armed and 25mW disarmed. Proof:

Armed
[image]

Disarmed
[image]
 
What VTX do you use? Did this work on older builds? And now it fails? In that case, can you please try to find out at what build and PR it started failing.

mhenrichsen, 2018-02-21:
Quick question. Does this feature differentiate on disarm and failsafe? 
Some people use the video to locate a quad after a failsafe.

AndersHoglund, 2018-02-21:
Good Q. don't really know. I tested with arming on a switch. And I needed the Tx on to see what's happening, if I switch Tx off for a failsafe test I would not see anything. Requires another test setup.

Gaz-B, 2018-02-21:
Right the two working machines are on 172c1e3 dated Jan 5th one is using unify pro HV the other is AKK X2 new firmware version. I have tried updating and rolled back to this version as it's been very stable.  The 3rd machine is on 7c36649 dated 19th feb unify pro 5V I have full OSD SA control but no auto power change on arm. PS I love that the AKK starts in pit mode goes to preset power on arm and backs down to 25mw on disarm.

AndersHoglund, 2018-02-21:
Sorry, I have no SA VTX, can not duplicate your findings. Either you yourself or someone else have to do some work and bisect versions to find the PR that causes this behaviour. All the build artefacts and version history you need for this are still available on: https://ci.betaflight.tech/job/Betaflight/changes

codecae, 2018-02-22:
I'll throw it on the bench and check it with a Unify (don't have an AKK).  Not sure what could've happened to the code to change the behavior.

codecae, 2018-02-22:
Using 3.3.0-RC2, I am seeing the expected results.

Armed -- second 0x01 byte = 200mw:
[image]

Disarmed -- 0x00 byte = 25mw:
[image]
 
OSD screenshots to follow

codecae, 2018-02-22:
Disarmed:
[image]

Armed:
[image]

codecae, 2018-02-22:
@Gaz-B I suspect the issue is that the firmware on the VTX may still be an issue.  Are you able to change channels using Lua scripts or the CLI?

Gaz-B, 2018-02-22:
I can change power via CLI and OSD but if low power when disarmed is off it never comes out of 25mw when this is active. This is on a genuine unify pro and I know the same issue will arise on other machines if I update to later firmware.

codecae, 2018-02-23:
I'm not sure how you're having this problem if you're using a unify.  The images above are visual proof of a 3.3.0-rc2 functioning properly with a Unify Pro.

codecae, 2018-02-23:
can you try placing the VTX channel OSD widget on the screen and see if the power index changes upon arming?

Gaz-B, 2018-02-23:
when I tested that's exactly what I did and with the function active it never exits power level 1.The only way to have more than 25mw is turn low power off in CLI. How ironic is it that this feature is my idea.

codecae, 2018-02-23:
How is the unify wired to the flight controller?

codecae, 2018-02-23:
uart or softserial?

AndersHoglund, 2018-02-23:
Do you really manage to Arm at all? 
I.e. if Arming Prevention is active, you can not arm and you will not get normal/high VTX power.

Easy to test, and I have verified this behaviour by doing this: 
1. Initial state, disarmed. VTX low power , 25mW
2. Raise throttle slightly above `min_check` ( To trigger Arming Prevention).
3. Flip the arm switch to Armed position. VTX still in low power mode.
This is exactly how it is intended to be by design.

Since Arming prevention was improved and sharpened at some point in version history, your tests and findings may have been done prior and after this change.

AndersHoglund, 2018-02-23:
@mhenrichsen While at it I also tested a failsafe situation, and that did also put the VTX in low power mode. Kind of expected, as failsafe puts the rig in disarmed state. At least in my config, not sure if there are other possibilies to configure failsafe.... But that would not be safe, and I have not investigated that any further.

So, if you want to use the video feed to find a lost and failsafed rig, do not use `vtx_low_power_disarm = ON` .

codecae, 2018-02-23:
There is question of whether or not this would be considered a bug... @AndersHoglund  your thoughts?

AndersHoglund, 2018-02-23:
Yeah,  I suppose VTX low power on disarm could be made conditional on failsafe.

AndersHoglund, 2018-02-23:
@Gaz-B can you confirm if your problem is related to Arming Prevention, please close this issue if that is the case.

Gaz-B, 2018-02-23:
I know it's arming as props spin and it flies. On the other machines to put back to high power without spinning props for surch purposes I use turtle mode.

AndersHoglund, 2018-02-23:
This is weird.  Is turtle mode involved also? Hmm, does that enable arming and override arm prevention? In that case we may be back to Arm Prevention again.

Since no-one can duplicate your issue, other than the Arm Prevention case, I think you must be much more specific what you see and how you see the problem, as well as how to reproduce it. We have had very little info so far to work with, only assumptions, guesswork and a lot of questions.  We have spent time trying to understand what it is you are seeing, now I think it is your turn to do some leg work. For example do a version bisect to find out exactly what PR causes this. 

Also, have you tried resetting your config to defaults, either by re-flashing with "Full chip erase" or the cli "defaults" command. Then re-entering all settings again from scratch on a clean base. There is also a "Reset Settings" button in the gui. It is a known problem to have strange behaviour when having a corrupt configuration, or a version mismatch between firmware and configuration.

Gaz-B, 2018-02-23:
I think most of the info I have given has been missed or misunderstood .I only mentioned turtle as the means on my other perfectly functioning machines to turn the VTX power back up after a crash without spinning the props it functions perfectly. I also stated on the working machines I tried updating to RC1 There abouts  and low power on disarm ceased working so I reverted to an early build of 3.3 and all was happy again I’m about to transfer all my config to that same older build to see if it works.

Gaz-B, 2018-02-23:
I have not flashed an earlier build onto it dropped a diff dump in with no cli changes and it's working and as I have said this is true of all my machines. anything post RC1 wont function.
  
# name
name Gt-R90

# resources
resource LED_STRIP 1 NONE
resource CAMERA_CONTROL 1 A00

# mixer

# servo

# servo mix

# feature
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL

# map
map TAER1234

# serial
serial 2 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1500 2100 0
aux 1 2 0 1950 2100 0
aux 2 13 0 900 1050 0
aux 3 19 2 1700 2075 0
aux 4 28 1 1500 2100 0
aux 5 30 2 1300 2100 0
aux 6 35 3 1700 2100 0

# adjrange
adjrange 0 0 2 900 2100 12 2

# rxrange

# vtx

# rxfail

# master
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set min_check = 1010
set max_check = 2000
set rssi_channel = 16
set rc_interp = MANUAL
set rc_interp_int = 14
set fpv_mix_degrees = 35
set min_throttle = 1050
set motor_pwm_protocol = DSHOT600
set current_meter = ADC
set vbat_scale = 111
set ibata_scale = 280
set ibata_offset = -2000
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 1
set yaw_deadband = 1
set pid_process_denom = 1
set osd_warnings = 63
set osd_cap_alarm = 400
set osd_alt_alarm = 122
set osd_vbat_pos = 44
set osd_rssi_pos = 2074
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2049
set osd_flymode_pos = 2070
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 2316
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 269
set osd_mah_drawn_pos = 2433
set osd_craft_name_pos = 2060
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 374
set osd_pid_roll_pos = 0
set osd_pid_pitch_pos = 0
set osd_pid_yaw_pos = 0
set osd_debug_pos = 1
set osd_power_pos = 33
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2249
set osd_avg_cell_voltage_pos = 2455
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 9
set osd_disarmed_pos = 2187
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 57
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set vtx_band = 3
set vtx_channel = 8
set vtx_power = 2
set vtx_low_power_disarm = ON
set vtx_freq = 5945
set vcd_video_system = 2

# profile
profile 0

set dterm_lowpass_type = PT1
set vbat_pid_gain = ON
set anti_gravity_gain = 3000
set setpoint_relax_ratio = 56
set dterm_setpoint_weight = 56
set p_pitch = 34
set i_pitch = 48
set d_pitch = 21
set p_roll = 28
set i_roll = 41
set d_roll = 18
set p_yaw = 25
set i_yaw = 35

# rateprofile
rateprofile 0

set roll_rc_rate = 127
set pitch_rc_rate = 127
set yaw_rc_rate = 120
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 75
set tpa_rate = 25
set tpa_breakpoint = 1600

Gaz-B, 2018-02-23:
I have just flashed RC2 #2049 and it's working but #2052 doesn't and I have lost track of the version I tried on another Quad that stopped the feature working it was just after RC1. I'm confused and happy that some current builds are playing nicely.

codecae, 2018-02-23:
I'll check a diff between those two commits.

AndersHoglund, 2018-02-23:
OK, we are getting close on the bisect, 
2049 OK
2050 ?
2051 ?
2052 Fail
Two more test and you may have found something. But I can not see what those builds has anything with VTX disarm power to do....

Gaz-B, 2018-02-24:
I hate to confuse this issue further. I just updated another machine with an omnibus F4 V5 to aec43a2 and RC2 and the SA didn't work on either so i Have rolled back again to 172c1e3 dated 5th Jan as it's functional.This is on an AKK X2 new batch VTX with revised smart audio. I can change settings  via OSD or CLI on all firmwares but once again it's the auto power change not functional on the latest ones.

codecae, 2018-02-24:
There seems to be a lot of variables in the testing.  Can we stick to one rig until we can narrow things down?

AndersHoglund, 2018-02-24:
Yes, was about to type a similar comment. Can we focus on one thing on one rig at a time please. It is very hard to follow what "...the two working machines..", "... the other machines.." and  "..updated another machine..."etc etc is all about. Please, a bit more specific and structured.

Gaz-B, 2018-02-24:
The first rig is now working on RC2. I was trying to move on with updating the rest of the fleet to RC2. I'm more trying to help you with things not being consistent with SA between builds by being informative

codecae, 2018-02-24:
If we could stick with hardware that is known to work beyond a shadow of a doubt, that would be helpful as well.  I do not personally have in my possession an "AKK X2 new batch VTX" so I can't perform adequate testing to find resolution.  There may very well be an issue with the VTX itself and chasing it down in BF will lead us nowhere.

codecae, 2018-02-24:
If you own a unify pro, please let's use it for testing because I can mirror your configuration.

Gaz-B, 2018-02-24:
The reason I have mentioned the other rigs to make it clear this feature stops working on all my machines on certain builds. This is why I don't believe it's hardware. As it works on RC2 on the Unify but not on the AKK this makes me wonder if a subtle change has stopped their new SA firmware  being compatible with later 3.3 builds.

Gaz-B, 2018-02-24:
OK so I just exactly the same test on am omnibus f4 V5 this time with a unify pro hv RC2 and latest build neither builds are auto power changing.
 
# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.3.0 Feb 18 2018 / 13:55:02 (f07bf40d1) MSP API: 1.37

# name
name Merica

# resources
resource LED_STRIP 1 NONE
resource CAMERA_CONTROL 1 B06

# mixer

# servo

# servo mix

# feature
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -MULTI_BEEPS
beeper -ARMED

# map
map TAER1234

# serial
serial 2 2048 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1500 2100 0
aux 1 2 0 1950 2100 0
aux 2 13 0 900 1050 0
aux 3 19 2 1700 2100 0
aux 4 28 1 1500 2100 0
aux 5 30 2 1300 2100 0
aux 6 35 3 1700 2100 0

# adjrange
adjrange 0 0 3 900 2100 12 3

# rxrange

# vtx

# rxfail
rxfail 4 s 1000

# master
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 0
set acc_trim_pitch = -2
set mag_hardware = NONE
set min_check = 1005
set max_check = 1998
set rssi_channel = 16
set rc_interp = MANUAL
set rc_interp_ch = RPY
set rc_interp_int = 14
set fpv_mix_degrees = 40
set serialrx_provider = SBUS
set airmode_start_throttle_percent = 35
set blackbox_p_ratio = 64
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 5
set failsafe_throttle_low_delay = 5
set align_board_roll = 180
set bat_capacity = 1200
set vbat_max_cell_voltage = 44
set vbat_min_cell_voltage = 30
set vbat_warning_cell_voltage = 32
set ibata_scale = 335
set ibata_offset = -2200
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set gyro_cal_on_first_arm = ON
set deadband = 1
set yaw_deadband = 1
set pid_process_denom = 1
set osd_cap_alarm = 1150
set osd_alt_alarm = 122
set osd_vbat_pos = 44
set osd_rssi_pos = 2105
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2049
set osd_flymode_pos = 25
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 2071
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 2433
set osd_mah_drawn_pos = 2402
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 2422
set osd_pid_roll_pos = 0
set osd_pid_pitch_pos = 0
set osd_pid_yaw_pos = 0
set osd_debug_pos = 1
set osd_power_pos = 2081
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2249
set osd_avg_cell_voltage_pos = 2455
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 9
set osd_disarmed_pos = 2187
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 58
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_min_rssi = OFF
set osd_stat_max_alt = ON
set osd_stat_endbatt = ON
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5917
set vcd_video_system = 2
set camera_control_key_delay = 125
set camera_control_internal_resistance = 99

# profile
profile 0

set dterm_lowpass_type = PT1
set anti_gravity_gain = 5000
set dterm_setpoint_weight = 100
set crash_delay = 40
set crash_recovery = ON
set i_pitch = 56
set p_roll = 56
set i_roll = 54
set d_roll = 33
set i_yaw = 48

# rateprofile
rateprofile 0

set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 146
set roll_srate = 81
set pitch_srate = 81
set yaw_srate = 77

codecae, 2018-02-24:
have you tried this on soft serial in 3.3?

Gaz-B, 2018-02-24:
The one working on 3.3 RC2 is a fury F4M FC on UART3 unify pro 5v. Both non workers on RC2 are omnibus F4 V5s also on UART3 one with unify pro hv the other AKK X2 all SA features working with the exception of auto power change.

codecae, 2018-02-24:
Just curious, are you leaving the usb cable plugged in when you test by chance or are you in the CLI?

Gaz-B, 2018-02-24:
Nope unplugged from usb then battery plugged in tested and power cycled just to be sure sitting with motors running for 15 secs or so. Also don't forget the version that works from the 5th jan never fails.

Gaz-B, 2018-02-25:
I have another anomaly the machine thought to be happy on RC2 won't auto power change to 800mw I can confirm in the CLI level 4 is selected. It works fine switching to 200 or 500 this is the Unify pro 5V set up. I thought it was working fine i only tested with 200mw so it must be borderline functional and the other two refuse completely on RC2.

Gaz-B, 2018-02-25:
All three machine now on 3.3 build from the 5th Jan and all working correctly.

AndersHoglund, 2018-02-25:
OK, now I think we are zooming in to something interesting. 

- Build from 5th Jan OK. 
- Next build 8th Jan Fails (Or ? )
- Failure only when trying to switch between 25mW and 800mW. (index 1 <-> 4(max))
- Ok when switching between 25mW and 200mW. ( 1 <-> 2 )

On the 8th of Jan there was a commit that might have something to do with this.  PR #4879 makes a change in vtx_common.c: 
```
-    if (vtxDevice && (index <= vtxDevice->capability.powerCount)) {
+    if (vtxDevice && (index < vtxDevice->capability.powerCount)) {

```
If SA power index is 4 and powerCount also 4, nothing will happen.
This does NOT affect my Tramp VTX, works fine to switch between 25mW (1) and highest 600mW (5). But there might be another situation with SA.  depending if `index` is 0-based or 1-based. This could differ between SA, Tramp and 6705. It is a mess with indexes like this being 0-based in some parts of the code and 1-based in other parts. Conversions are made everywhere, both +1 and -1 etc. Easy to make mistakes here. I have not traced what value range is used with SA here, will try to do that. Meanwhile, you guys with SA could revert this line of code and test it.

codecae, 2018-02-25:
Good eyes!  Sneaky little indexing changes like this are hard to spot.  I'll see if I can recreate this @ 800mw

Gaz-B, 2018-02-25:
Yay progress hopefully this will address the Omnibus machines that still don't auto power change at all on the later builds.

AndersHoglund, 2018-02-25:
Yepp, there is a total mess when it comes to defining power index base, value range and if the number of power levels should include 0-power (power off) or not. Tramp, rtc6705 and SA are all doing this differently. 
SA do not support power off, but assumes that vtx_common do support power off. But SA do not include power off in the number of power states supported. That is a missmatch, or? What a mess.
Reverting the line in prev. post will probably be the easy fix and q&d patch. A total overhaul would be the correct way. Will do a PR on the easy fix.

AndersHoglund, 2018-02-25:
But still, how is it possible to set highest SA power level via cli and CMS? That should also fail in that case. Does not really make sense.

AndersHoglund, 2018-02-25:
> Yay progress hopefully this will address the Omnibus machines that still don't auto power change at all on the later builds.

@Gaz-B Is this tested with 200mW or 800mW or both?

codecae, 2018-02-25:
@AndersHoglund Agree with the overhaul -- i'm currently working on building an entirely new vtx subsystem that abstracts this into a more unified experience (one set of CMS screens, custom power tables instead of predefined band/chan conversions, etc.)

Gaz-B, 2018-02-25:
I cant remember now I may have only tested the omnibus at 800 that would explain the results of them apparently being different on RC2 to the other unit. that at that point had only been tested at 200.

Gaz-B, 2018-02-25:
I just checked the diff one of the machines I was testing that has not been flown today and yes yes power level set to 4.

AndersHoglund, 2018-02-25:
OK, should we just revert the change in PR 5879 for now as a quick fix for BF3.3, a gap filler while waiting for a proper solution later on? Has that reversal been tested?

AndersHoglund, 2018-02-25:
@Gaz-B can you build and test PR #5313 please. If you have problems building, please let me known what FC targets you need and I can build them for you. I have seen you use OMNIBUS and OMNIBUSF4SD , anything else?

AndersHoglund, 2018-02-25:
@Gaz-B for your testing: [betaflight_3.3.0_pr5313.zip](https://github.com/betaflight/betaflight/files/1756625/betaflight_3.3.0_pr5313.zip)

Gaz-B, 2018-02-25:
I have OmnibusF4sd, OmnibusF7 V1, KiwiF4 V2, furyF4OSD,

Gaz-B, 2018-02-25:
We have a winner flashed it to the omnibusf4sd machine with the akk and it works.

AndersHoglund, 2018-02-25:
OK, good news.  Some more targets in here: 
[betaflight_3.3.0_pr5313-2.zip](https://github.com/betaflight/betaflight/files/1756766/betaflight_3.3.0_pr5313-2.zip)

Gaz-B, 2018-02-25:
Thats 2 for 2 the other omnibusf4sd with unify ok

Gaz-B, 2018-02-25:
Fury is happy too that's all 3 complete serviceable machines working.

Gaz-B, 2018-02-25:
Is this a Modified version of this build #2066 (Feb 25, 2018 1:54:08 PM) I like to know what i'm running.

mikeller, 2018-02-25:
@AndersHoglund, @codecae: Another complicating factor in the mess that is power level selection is that RTC6705 hardware exists in 2 different 'flavours', 0 based and 1 based, depending on the hardware pin to disable the VTX output is connected or not. :-(

[3 more replies]