betaflight/betaflight #5261: BLHeli_32 passthru does not work on RTFQ Omnibus F4 w/Robotdog ESCs

aowi7280, 2018-02-21:
Passthru did not work for 2 ESCs. I tried to reflash several times and it always failed. I was able to change/save settings on the 2 good ESCs. So I got replacements. I installed the 2 new ESCs and they also do not communicate properly. And now one of the other ESCs is no longer recognized by BLHeli_32. So 3 ESCs do not communicate, 1 does communicate. They run OK and I can fly, but I cannot change any of the settings because BLHeli_32 does not recognize them.

I was using Betaflight 3.2.4 in my flight controller. Then I upgraded to Betaflight 3.2.5 and the BLHeli_32 pass through still does not work right; using latest revision. I even tried the BLHeli Beta version, no good. Tried different USB ports on PC, no good.

etracer65, 2018-02-21:
This is something you probably should take to the BLHeli_32 rcgroups forum, but one thing I've found is that some BLHeli_32 esc's won't communicate properly over passthrough if your motor protocol is DSHOT.

Try temporarily switching your ESC/Motor Protocol to ONESHOT125.  Then disconnect from the configurator and try connecting via BLHeli suite.  Don't forget to change your motor protocol back when you're done.  You'll also need to reset your PID loop frequency because it will have changed to 2KHz for ONESHOT125 support.

aowi7280, 2018-02-22:
Thanks for the suggestion. It did not work, though.

closed 2018-02-22 by mikeller

richard-scott, 2018-02-22:
Also maybe swap the signal wire on the working ESC's with the wires of the non responsive ESC's. You will then see if it is the FC or the ESC at fault.