betaflight/betaflight #5263: Please add feature to reduce Tricopter Tail-Wag.

ResinSmoker, 2018-02-21:
# A - Description

Would it be possible to add some sort of Servo or Yaw specific "smoothing" so Tailwag wouldn't be so intense. Often the Wag is out of control making the Tricopter either unflyable or near impossible to tune.
It's as if the Servo is constantly overshooting target regardless of Filter / PID / CLI settings used.

# B - Steps To Repeat

Typically high frequency Wag is seen immediately after arming, before throttle up. After which the Wag dissipates while the throttle is being manually held above idle. Later low frequency Wag can often show up during sharp turns and fast straits. This makes the Yaw control sloppy at best, unflyable at its worst.
Lowering the Yaw P-Term to extremely low values does remove the pre-throttle up oscillation but essentially makes the Copter unflyable. Lowering the tri_dynamic_yaw_boost can also help but it gets to the point where it makes the Tricopter unflyable as well.

Note: Full Tail-Tune / calibration completed.

Edit: The Wag appears to be a combination of both Yaw (naturally) and Roll. With the Roll being induced from the motors mass being held slightly above the Servo's pivot point. In a sense the Copters weight is shifting from side to side slightly as the Servo pivots its axis, tossing the motor to one side or the other. There are a large number of copters that handle just fine like this but others, especially those with the more sensitive IMU's tend to overreact badly.

# C - Video Links

Examples:
Stationary wag https://www.youtube.com/watch?v=SeXPp4LuZms
Under Throttle https://www.youtube.com/watch?v=f6tYqGBGsXg

# D - Forum post links

There are tons of links as this is a very common issue for Tricopters.

# E - Version Information

Betaflight 10.1.0
Triflight on Betaflight 3.2.2 v2 / MATEKF405 (MKF4) 3.2.2 Dec 31 2017 / 15:24:22 (30f68c8) MSP API: 1.36

# E - Other information

-- Flight controller make/model : MATEK F405 AIO
-- Dump: https://pastebin.com/x3118UF8
-- Diff: https://pastebin.com/0YJesKyN
-- “Aikon” AK32 35A 6S BLHeli_32 ESC’s See image for more ESC details https://flic.kr/p/DXf5Kp
-- FrSky R-XSR with EU firmware from mfr (FW-RXSR-171103)
-- Taranis X9D (EU) FW: opentx-x9d+ Vers: 2.2.1 Date: 2017-12-17 EEPR: 218
-- Schematics (None)
-- Wiring diagrams: https://flic.kr/p/23sQEuz
-- Windows 10
-- Tricopter wired up with 470uf 35v capacitors for all ESC's and 1000uf 35v for Battery.
-- Servo has attached feedback wire.
-- Everything is soft mounted, except for the Sevro and Tilt mechanism (not possible).
-- Also have an F3FC based Tricopter from RCexplorer with the same issues.
-- [Blackbox of Tricopter Tail-Wag 2-21-2018.zip](https://github.com/betaflight/betaflight/files/1745934/Blackbox.of.Tricopter.Tail-Wag.2-21-2018.zip)
If for some reason it does not show try using my Google Drive: https://drive.google.com/file/d/1tUX49k2bRePZNE_VHBLZVh-HK8wEIy4f/view?usp=sharing
-- Soon to upgrade the FC to Matek F405 CTR with the less sensitive IMU. Once that's done I'll report back with results.

mikeller, 2018-02-21:
@ikaino, any thoughts?

DieHertz, 2018-02-22:
@lkaino

jflyper, 2018-02-22:
Isn't this correctable by PID tuning???

mikeller, 2018-02-22:
@jflyper: It is certainly tunable. But it's a trade off between ability to corner hard and probability of a wobble developing. Depending on the wobbliness of the mechanical construction of the tail, the tendency of wobble is always there. On my cheap plastic Trifecta I sometimes get it when landed / disarmed.

ResinSmoker, 2018-02-22:
For some Tricopters there is the possibility to "tune" some of the Wag out but not all. Like Mikeller stated, its a trade-off and a rather big one at that. However there are some builds that no amount of tuning can make for a flyable experience and this is with all new hardware.

IMO what's going on to entirely due to the relationship between the Copters IMU and Tail Mechanism feeding off / reacting to each other. Unfortunately this is not something we can filter out with the current methods.

jflyper, 2018-02-22:
It seems to me that there is a mechanical (structural) problem with the tail section, not to speak of the hanging wire swinging in phase.

ResinSmoker, 2018-02-22:
Considering how the system works that's to be expected. All a builder can do is attempt to minimize the structural issues. What is needed is some way for the software to compensate for the expected swing in mass creating the undesired roll / yaw counter reaction. At the very least a way to slow the reaction down some "smoot" when Wag is apparent.

jflyper, 2018-02-22:
Ah... I'm beginning to learn something new (again, for today). TBH, I was bit surprised to see the wobbling this bad, as my Baby-Tri never does it. We sure need a yaw control that knows what happens when it gives a servo command.

ResinSmoker, 2018-02-22:
Can you link the Ikaino's lecture please?

jflyper, 2018-02-22:
Ah, no, keep screaming for him :)

ResinSmoker, 2018-02-22:
@lkaino, when you have a moment, we'd would love your input.

Thanks in advance!

mikeller, 2018-02-23:
Hehe, I bet @lkaino's ears are ringing by now. :-P

ResinSmoker, 2018-02-23:
Playing around with this more I've discovered that lowering the Gyro Update / PID loop frequency does appear to help some. Granted, this does have the unintended side effect of reducing the Copter responsiveness. Obviously there's a limit to how low of a setting would be desirable with respect to the stability gained. Along this same line of thinking, would it be possible to adjust the Gyro's Update / PID loop frequency for each individual axis?

Another other idea explored was increasing the base idle speed from 4.5% (D-shot) to 10% in incremental steps. Reasoning behind this is that WAG stops while manually holding the throttle above idle. Unfortunately, the same can not be true of just increasing the idle speed alone as apparently there is another mechanism at work here that is that only comes into play when actively throttling.

ResinSmoker, 2018-02-27:
"Hehe, I bet @lkaino's ears are ringing by now. :-P"

I guess not...