betaflight/betaflight #5265: BETAFLIGHTF7 target

flash-v, 2018-02-21:
Does anyone know target for this board? 
https://www.rcgroups.com/forums/showthread.php?3024138-Betaflight-F7-RV1-FC-The-one-to-rule-them-all-)
https://www.fpvmodel.com/betaflightf7-rv1-flight-controller_g1393.html

OMNIBUSF7 (or OMNIBUSF7 V2) not work for this or work incorrectly.

McGiverGim, 2018-02-22:
Maybe this? https://github.com/betaflight/betaflight/pull/5241

DieHertz, 2018-02-22:
It should be just a clone of Omnibus, weird that blackbox doesn't work. Have you formatted your SDcard?

flash-v, 2018-02-22:
DieHertz, This board does not contain a SDcard. The black box is made on a flash memory chip. If I knew how to recompile the omnibus firmware to work with flash memory, I would gladly do it.

flash-v, 2018-02-22:
**McGiverGim**, yes, you're right

DieHertz, 2018-02-22:
@flash-v sweet, I'll unpack mine tonight and try to help you if @rotcehdnih won't beat me to it :-) looks like BFF7 requires a custom target after all

flash-v, 2018-02-22:
**DieHertz**, many thanks. If you still tell me how can compile my own target for board, I will be grateful. :)

DieHertz, 2018-02-22:
Unfortunately the existing one isn't fully functional, but there's a guide how to build BF in our wiki

flash-v, 2018-02-22:
I think this issue can be closed.

closed 2018-02-22 by flash-v