betaflight/betaflight #5271: 3.3.0 RC2 on Matek 405-STD no connection

antikris84, 2018-02-22:
After Flashing 3.3.0 RC2 on the Matek 405-STD. It will not reconnect to betaflight configurator.
The same issue appeared on RC1.

Rolling back to 2.5.0 works fine tho.

mikeller, 2018-02-23:
Tried with a MATEKF405-OSD (essentially an older version of the MATEKF405-STD). All working fine.

Have you tried 'Full chip erase' when flashing?

antikris84, 2018-02-23:
Yes I tried that just now, mostly because I did not remember, but the
result was the same.

It reports the following in the status line
2018-02-23 @ 09:22:22 -- Serial port successfully opened with ID: 3
2018-02-23 @ 09:22:32 -- Serial port successfully closed

fre. 23. feb. 2018 kl. 08:37 skrev Michael Keller <notifications@github.com

>:

> Tried with a MATEKF405-OSD (essentially an older version of the
> MATEKF405-STD). All working fine.
>
> Have you tried 'Full chip erase' when flashing?
-- 
MVH

Kristian Johansen

codecae, 2018-02-23:
Not to sound dismissive, butI have two of these boards and have not run into this issue.  Can you try with 3.2.5?

jflyper, 2018-02-23:
Bare NUCLEO-F405RG runs 3.3.0-RC2, 3.3.0-RC1 and 3.2.5 without a problem.
I conclude there is a problem with flashing or some hardware oriented problem with the board under question. Should consult the board distributor.

jflyper, 2018-02-23:
Make sure Full Chip Erase before flashing...

antikris84, 2018-02-23:
Sound like it's just my board then. 
3.2.5 works fine. It may be the FC-hub. Will try to disconnect and try with just the FC alone.

antikris84, 2018-02-23:
Tried Without the FC-hub VTX witch i found out was defective. And removed the SD card for good measure. And now it is working fine.

closed 2018-02-23 by antikris84

jflyper, 2018-02-24:
I'm interested in how the FC-HUB VTX has failed the RCs to fail. It's nice if you can add something when you have a spare time @antikris84 😉

codecae, 2018-02-24:
Doesn't that hub use a ribbon cable?

antikris84, 2018-02-24:
Looks like i have tried to many things at once. After a little experimenting I found that it was the SD card. I have to format the SD card after installing 3.3 and then connect to the computer. The FC hub did not make it fail.

codecae, 2018-02-24:
Did you also possibly re-seat the ribbon cable when you swapped things out?  I'm wondering if maybe something got shorted out by means of a slightly crooked ribbon cable.

antikris84, 2018-02-24:
No it shorted out because I crashed in snow. Everything is Conformal coated. But the ribbon cable make it hard to be 100% sure. And the coating makes it hard to remove the cable, so I will replace it before connecting it to the new hub witch i will not have for another 3 weeks.