betaflight/betaflight #5272: Discussion: Custom ledstrip blink patterns options

etracer65, 2018-02-22:
Looking for feedback from the developers.

I was looking to implement a way to allow the user to specify a custom blink pattern for the ledstrip.  The existing pattern (2 quick blinks, long pause, repeat) is not very useful as a racing taillight as the blink pattern is very slow.  I prefer a fast blinking as it's much more visible.

So there are two approaches:

1. Allow the user to specify something completely custom by defining the full 16bit bit pattern that indicates the on/off cycles of the LED.

2. Predefine a set of patterns and let the user select one via an index

I originally wanted to go with (1) but ran into a few problems.  First the `.config.minmax` validation is `int16` so trying to specify a full uint16 range results in overflow warnings.  So changes would need to be made to support uint16 validation.

Next, I'm not sure the UI for a user to specify the pattern will be understandable by most.  For example if they wanted a simple fast blinking the pattern would be `0b0101010101010101` but the user would need to enter decimal 21845 in the CLI.  So I'm not sure this is viable unless we implemented a "custom blink editor" in the Configurator that's easy to use and this may be overkill for this small feature.

Option (2) is simpler to implement but of course has less flexibility.  It would require coming up with a set of "useful" blink patterns and documenting them.  Probably open the door for requests for additional patterns, but there's only ever been one to this point (which I suspect isn't used much because it's not very useful IMO).

Thoughts?

mikeller, 2018-02-23:
I'd go for 1.

- a lot of the LED configuration is already done by bitmaps, and the CLI getters / setters are only really useful for backup / restore (goes for a lot of non-LED settings as well);
- to be used as a getter / setter, wasting 1 bit, or using the `int16` range for this settings does not seem to be a crippling limitation.

etracer65, 2018-02-23:
I like option (1) as well, but there are a few side enhancements that would be needed to make it work.  The actual storage of the 16bit bit pattern isn't an issue.  Rather it's that there's currently no way to define a CLI parameter as an unsigned 16bit number.  And we need the full 16 bits (or at least an even number of bits) to create smoothly repeating patterns.  It would be even better to use 32bits but that's overkill.

For example:
`    { "ledstrip_blink_pattern",     VAR_UINT16  | MASTER_VALUE, .config.minmax = { 0, 65535 }, PG_LED_STRIP_CONFIG, offsetof(ledStripConfig_t, ledstrip_blink_pattern) },
`

gives compilation warnings because `.config.minmax` works on int16 so expects -32768 to 32767

`./src/main/interface/settings.c:725:82: warning: overflow in implicit constant conversion [-Woverflow]`

And it also misbehaves in the CLI validation:

```
# get ledstrip_blink_pattern
ledstrip_blink_pattern = 5
Allowed range: 0 - -1

# set ledstrip_blink_pattern = 21845
Invalid value
Allowed range: 0 - -1
```

In earlier discussions on Slack @DieHertz Thought this was something he could fix and add full uint16 support in the validation.

Then I guess sometime later we could add a "blink pattern editor" in the Configurator to help people build the 16bit pattern - assuming someone gets the spare time. :)

ledvinap, 2018-02-23:
@etracer65 : SOS is 22 bits minimum ;-)

I have implemented 'bytecode' interpreter for (single-color,single led) blinking for one project ... with ramp support and button synchronization, all running on AtTiny13 (1k flash) ... it has 4 instructions (JMP, BTN, SET, CMD) and simple assembler, people with Art background learned it surprisingly fast ...

etracer65, 2018-02-23:
> SOS is 22 bits minimum ;-)

Yeah, I know.  But I figured 16 bits was as much as we could expect people to be able to deal with.  Going to 32bits would require people to use 1431655765 for a fast (5Hz) blink - not very intuitive (at least for non-programmers). :)

I planned on publishing a list of standard patterns and their representative decimal numbers in the ledstrip docs to help people out.

Maybe we could enhance the CLI to accept entering binary number representations - 0b010101...  to make it more intuitive to use.  Or at least hex - 0x5555, etc.

ledvinap, 2018-02-23:
Just replace `atoi` with `strol` to get `0x` prefix support ...

ledvinap, 2018-02-23:
@etracer65 : Also, with 32 bits, it may be good idea to use highest bit as terminator, repeating cycle immediately (shifting down, reload when value is 0x01). This will remove problem with sequences that don't fit/wrap in 32 bits
0x1 will be all off, 0x3 all on, 0x5 fast blink

etracer65, 2018-02-23:
Yes, that would give the most flexibility (and actually how the pattern currently works), but trying to balance between capability and making it understandable for non-programmers is the question.  Describing a 16bit pattern where the blinks are "played" by shifting to the lsb until reaching a stop bit and then the pattern repeats...

I'm fine with making this a "pro" feature for now as it's not something that people can currently do anyway.  If you understand the logic then go for it.  Otherwise use one of the provided sequences.  This way I also could change to 32bit patterns using the stop bit structure.  Then look at improving the UI down the road (maybe configurator support).

Still need to fix the CLI to accept unsigned 16bit (or maybe 32bit) numbers though.

ledvinap, 2018-02-23:
@etracer65 : CLI supports byte array, it should do fine for bitmap.

Also, with 32 bits, just tell users to prepend 0x1 before (4bit-rounded) bit pattern

etracer65, 2018-05-05:
@DieHertz Do you have any timeline to add the uint16 range validation for settings that you demoed on slack?  I need to be able to add a parameter using the full 16bit (0-65535) range to move this forward.

mikeller, 2018-05-05:
@etracer65: Have you considered using an array instead?

etracer65, 2018-05-05:
I could make it work with a byte array, but the UX and explaining to the user how to build the bit pattern is already complicated enough.  Then adding on needing to split across multiple bytes, byte ordering, etc.  I can go that route if needed but it appeared that the uint16 validation was close so I'd rather wait for that if possible.

closed 2026-09-26 by blckmn