betaflight/betaflight #5275: BUG: "3D on a switch" broken in 3.3.0 RC2

thongraegu, 2018-02-22:
After this PR feature doesnt work anymore https://github.com/betaflight/betaflight/pull/5179

"switched_3d = off:
BOX3D switches between 'normal' 3D (idle mid stick) and non-3D.
switched_3d = on:
BOX3D switches between normal and inverted 3D."

Thats broken, now when "switched_3d = on" using the switch puts motors full throttle on reverse when I have minimum throttle on my radio instead of usual minimum throttle reversed.

mikeller, 2018-02-23:
Tried with a MATEKF405-OSD (essentially an older version of the MATEKF405-STD). All working fine.

Have you tried 'Full chip erase' when flashing?

thongraegu, 2018-02-23:
I tested again RC1 vs RC2 both clean fresh install with "Full chip erase" on. For RC2 I have CLI command switched_3d=on enabled.

I filmed what happens with motors.
https://youtu.be/seV5oj4LDrY

etracer65, 2018-02-23:
@thongraegu I have implemented a fix for this problem.  Can you give https://github.com/betaflight/betaflight/pull/5291 a try and report back?  If you can't compile the code yourself, let me know what target you need and I'll make you a build to test.

I don't have a 3D quad to test but I verified all of the motor outputs behave the same as BF3.3RC1 so I think this should fix it.

thongraegu, 2018-02-23:
@etracer65 bff4 target please

etracer65, 2018-02-23:
Ok, but it’ll be a couple of hours before I get back home.

mikeller, 2018-02-24:
@thongraegu: Your order is coming up on http://ci.betaflight.tech/ in about 30 minutes. :smiley:

closed 2018-02-24 by mikeller