betaflight/betaflight #5283: Max RPM in OSD statistics screen and ESC telemetry data recorded in blackbox. (Feature Request)

BludzUK, 2018-02-22:
Please can we add 'Max RPM' to the end screen statistics. I really like to see the difference between props motors and batteries and this would be very helpful.

Having all the esc data in blackbox would also be nice. Individual amp draw, esc and voltage for each esc in blackbox would be good for diagnosing issues and/or improving frame design and setup.

mikeller, 2018-02-23:
Max RPM in statistics is something that can be done.

Have a look at https://github.com/betaflight/betaflight/blob/9d7c8fbd46d6519b86d56ffd78da5acbccbd89ad/src/main/build/debug.c#L31-L59, there is already a lot of things that are reported. One of the limitations of blackbox is that it can only report up to 4 custom values in addition to the default values that are always logged. Increasing this is not simple, since it will mean increasing the bandwidth of data that is logged, forcing the maximum logging rate to be slower.

BludzUK, 2018-02-23:
Thanks Mike. Glad to hear the 'max rpm stat' can be added.

Didn't see the debug mode option for esc-sensor, I'll give that a go. Blackbox seems to really struggle with all the data and faster speeds. What would be a logical step to improve blackbox? Is it just a hardware limitation with the baud rate and a better chip would improve this? Or does it need a bit of a rewrite to clean it up.

I'm still learning how the source code works and have been playing with some osd stuff.  Quite a bit to get my head round but I'm getting there. The changes that are made to add max rpm will help me learn too, soI'm looking forward to that.

Keep up the great work mate.

mikeller, 2018-02-23:
@BludzUK: It's mostly a hardware limitation. On-board SPI flash seems to be able to cope well with the speed, but it's normally limited to only a couple of MB, which amounts to a short flight time at full speed, and downloading it is a hassle so far (adding a MSD driver for it would be a great improvement).
As for SD card, replacing the current SPI configuration with SDIO might bring an improvement, we have an implementation, but there's pretty much no boards available that support it.

BludzUK, 2018-02-23:
I agree, onboard spi is annoying to download,

Seems like this has been a bit of a bottleneck in the extensive Betaflight toolbox for a while now, but because default pids work for so many people ye olde blackbox isn't needed by most. I'm developing a stretch plus frame and it's really very useful for me.

Maybe with the new filter people will actually want to use it again.... then again so many people loving the new filter without it even turning it on and tuning it, so who the hell knows, lol.

mikeller, 2018-02-24:
@BludzUK: I think blackbox is still an important tool for everybody wanting to tune his craft - be it PIDs or, today probably more common, filters.
There have been some gradual improvements to the download speeds, with MSP jumbo frames and huffman encoding for the data, but nothing revolutionary unfortunately.