betaflight/betaflight #5292: No voltage display from blheli 32bit ESC telemetry

kwadyote, 2018-02-23:
I'm using 4 xracer 25A blheli 32bit ESCs and although I can see the individual telemetry voltages from each ESC in betaflight configurator in the battery tab, it does not report an average of the voltages to the FC but instead reports 0V.  Using a FF Fortini F4 OSD FC.

jflyper, 2018-02-24:
Do you have all four voltage readings?

kwadyote, 2018-02-25:
Yes I do.

jflyper, 2018-02-26:
@kwadyote Whatever the version you are running, can you do CLI `defaults`, and immediately configure motor/esc protocol and battery meter configuration only, to see if the issue persists?

kwadyote, 2018-02-26:
I will try tonight.  Using v3.3.0 RC 2

kwadyote, 2018-03-07:
The reset looks to have fixed the issue.  Below is the diff before reset:
# diff

# version
# Betaflight / FF_FORTINIF4 (FORT) 3.3.0 Feb 18 2018 / 13:39:35 (f07bf40d1) MSP API: 1.37

# name
name Kwadyote

# resources
resource LED_STRIP 1 NONE

# mixer

# servo

# servo mix

# feature
feature AIRMODE
feature ESC_SENSOR
feature DYNAMIC_FILTER

# beeper
beeper -ON_USB

# map
map TAER1234

# serial
serial 0 1024 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0
aux 1 13 4 1700 2100 0
aux 2 35 1 900 1300 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_notch1_hz = 0
set rssi_channel = 16
set blackbox_device = SPIFLASH
set motor_pwm_protocol = DSHOT1200
set bat_capacity = 1300
set current_meter = ESC
set yaw_motors_reversed = ON
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warnings = 63
set osd_vbat_pos = 2433
set osd_rssi_pos = 353
set osd_tim_1_pos = 407
set osd_tim_2_pos = 2423
set osd_vtx_channel_pos = 2071
set osd_current_pos = 2048
set osd_mah_drawn_pos = 2080
set osd_craft_name_pos = 2453
set osd_esc_tmp_pos = 2105
set osd_esc_rpm_pos = 160
set vtx_band = 5
set vtx_freq = 5658
set vcd_video_system = 2
set camera_control_key_delay = 100
set camera_control_internal_resistance = 99

# profile
profile 0

set dterm_lowpass_type = PT1

# rateprofile
rateprofile 0

closed 2018-03-12 by kwadyote