betaflight/betaflight #5295: Flip32F3Omnibus OSD USB Com port issue

hyptx, 2018-02-24:
1. I have no video on vout pin, but camera transmits fine when connected directly. So I tried to update the firmware but the flip32f3osd firmware bricks the FC 3.2.5. Does not recognize the board when plugged into the computer. 

I opted to try to flash the Omnibus 3.2.5 firmware, now I can connect to the board, but still no osd, or video, just black screen. There is a good chance that the osd chip is bad. I am only submitting this because the firmware to match the board breaks it.

mikeller, 2018-02-24:
@hyptx: Can you try the 3.3.0-RC3 release of the FLIP32F3OSD target?

hyptx, 2018-02-24:
Nogo, I did not see RC3 as an option but tried RC2. The board wasn't recognized. I flashed the OMNIBUS target back and the board is working again but no osd. I think I just have a bum osd on the board: http://www.readytoflyquads.com/flip-32-f3-omnibus-bec-w-baro-option

mikeller, 2018-02-24:
@hyptx: I assumed the OSD was borked - video passthrough is entirely analogue circuitry, and nothing that a firmware update could fix.
Re the board wedging, that's odd. Does it work with any version of the `FLIP32F3OSD` target, or could it be that you just have a mislabelled clone?

jflyper, 2018-02-24:
As I've commented in the RCG's Omnibus F3 thread, FLIP32-F3-OSD is SPRacing F3 + MinimOSD, and would never work with OMNIBUSF3 target. First and foremost, SPRacing F3 has CP210x chip and Omnibus uses VCP. If you can connect to your board with OMNIBUS target flashed, then the board can not be FLIP32-F3-OSD.

jflyper, 2018-02-24:
Since this is not a firmware issue, please continue to use the RCG thread for support.

closed 2018-02-24 by jflyper