betaflight/betaflight #5297: Betaflight 3.3 RC2 Omnibus - Motor Stop functions not working

robthomson, 2018-02-24:
When testing betaflight 3.3 rc2 on an omnibus board, I found the motor stop functions did not work.

I normally fly with my motors only disarming once I reach zero throttle.  This gives a nice controlled landing and motor cut on touch down.

With 3.3rc2 I found that the changes in the gui whilst saved to the quad seem to have no impact while in flight.  Essentially the disarm switch causes an instant motor cut.

I then moved on to trying the new 'timed delay' feature for the motor cut.  This seemed to have no effect either, with the motor cutting out as soon as you disarm.

mikeller, 2018-02-24:
@robthomson: Please read the release notes - the fact that this has been removed is listed in there. I don't know what 'timed delay' you are referring to, where / how do you configure it? (Also, note that immediately stopping the motors when the disarm switch is activated is an important safety feature.)

robschii, 2018-02-27:
mikeller  maybe you have to think about it again.  Again agian agian again.

http://butterflight.co/

wind0r, 2018-02-27:
@robschii such message from you isnt helping here at all.

I removed that feature because i think it is a safety concern since i know some people which had problems with that "feature" and got hurt.

First step was to make that setting hidden under the experts tab and then since we do not have any usage statistics and I thought no one is as crazy and is using that feature i removed it from the firmware in 3.3. Other people agreed with that the saftey of the users is more important then a dangorus feature that even could implemented in the radio itself.

If there are other Opinions about that people need to sanely talk about that and not stat pointing fingers and such.

I personlly still think it is dangerous and i wouldnt reimplement it as a feature for just 2 persons missing it.. If there are more persons that may could be reinteduced as a cli setting.

robthomson, 2018-02-27:
I personally find it is fantastic - but agree it required discipline to use it.

It allows you to very easily land in a controlled way; for me and pretty much all the guys I know, its the way we all operate. (nobody has ever had an issue in this way - but I can see how it could go wrong)

If it could be kept as a cli option I would be happy with that?

That being said.. yes. The radio could implement the same function if required.  Guess the main thing is to know if it is or is not going to be available so I can update my model configs to suit.

Rob

etracer65, 2018-02-27:
An alternate way to achieve the same result without the safety issues is to:

- Enable `motor_stop`
- Disable feature airmode (so it can be set up on a switch)
- Configure your arming on a 3-pos switch so that you have disarmed -> armed -> armed + airmode

Now when you come in to land, switch to the middle setting to disable airmode.  This will then enable motor_stop.  Simply land and lower the throttle to zero and the motors stop.  Then fully disarm.  Same end result with fewer safety issues.

If you don't want to use a 3-pos switch, simply set up a "landing mode" switch that does the airmode on/off.

robthomson, 2018-02-27:
Lol.. Pretty much how i do mine.

mikeller, 2018-02-27:
@robschii: What did you want to say? Of course you are free to switch to firmware that is less safe than Betaflight, it is up to you how much risk of hurting yourself or others you want to take.

closed 2018-02-27 by mikeller

robthomson, 2018-03-01:
Incase anyone wants to know the logical switch setup to replicate this function on Opentx.

[image]

Simply set your arm switch to L1

Rob

cherepuga, 2018-03-02:
@wind0r  Just want to vote for a possible re-introduction of that feature.  I use it and so do my friends.  We never had any problems and in an emergency i trained myself to disarm and flip throttle to zero almost simultaneously.  

Either way, you are doing a phenomenal job and i can always implement a workaround.  Thank you.

oleonav, 2018-03-03:
@wind0r, I agree with @cherepuga , having at a minimum the cli option available would be great

halfcab1981, 2018-03-06:
Another vote here for implementing it in the CLI please. I understand completely why it is removed, but a newcomer to betaflight would need to go very far astray to accidentally turn this on if it is in the CLI. Definitely keep it out of the configurator though.

And thanks for the good work devs.

bulby97, 2018-03-07:
Man I have been through a hard crash not knowing this. Keep that feature and just add a huge warning, or expert mode, but common guys. 

Look at that website : https://oscarliang.com/avoid-accidentally-disarm-mid-flight/

This is where I found the trick arm + airmode, and a LOT of people are using this

floydhelios, 2018-03-13:
I installed 3.3 and crashed because disarm_kill_switch is gone. I used this feature on all my quads and I'm used to it, so when I flipped the switch prior to coming in for a landing my machine dropped (luckily only broke a prop). Please bring disarm_kill_switch back! Leave it off by default and keep the option out of the GUI to prevent folks from accidentally setting it to 0 if the worry is that it's unsafe. But leave it in the CLI so those of us that want it can use it. I personally have never had a problem!

alancastell, 2018-05-11:
I agree with having it at least in CLI.  Have used this feature for years now and it makes more sense to pre disarm so you you just have to drop throttle to 0 to land.    Having to drop throttle AND flip the switch can lead to some bouncy landings.  Put it somewhere that a newbie couldn't turn on by accident and accept that as being just as safe as taking it out.

SangaeLama, 2018-06-21:
Yeah It should atleast be in CLI as me and my friends are also used to this feature. We used to hit the disarm switch prior to landing and then in actual landing we just needed to ground the quad, it was an awesome feature, all of my friends (including me) crashed out quads as we were used to that feature.
Please bring it back, people not using it wont even notice it in CLI.
Thank you if you hear me... :)