betaflight/betaflight #5298: Telemetry/SBUS works only when Connected to BF Revolt Target 3.2.5

mawais15, 2018-02-24:
Hi,
I have a strange issue with Revolt V1 BF 3.2.5 target. The sbus and telemetry works only when USB is plugged in and FC is connected to BF. I am using Frsky X4r with inverted telemetry hack on UART1, telemetry on UART3 and TBS on UART6. Quad does not arm and telemetry does not work when not connected.

jflyper, 2018-02-24:
Is your receiver powered when NOT connected to USB?

mawais15, 2018-02-24:
i am not sure, receiver is wrapped up in tape however i do not get all starting tones of motors (like you normally do) i just get three consecutive beeps followed by none (in normal u get two more long beeps)

mawais15, 2018-02-24:
receiver gets power. I just confirmed by unwrapping it. Also rssi comes in taranis which means its getting power

mawais15, 2018-02-24:
It was a hardware problem, my revolt V1 is having issue with initializing sometimes. Sorry to bring this up without digging!

closed 2018-02-24 by mawais15