betaflight/betaflight #5301: Betaflight 3.3-RC2 gyro drift a lot and Accelerometer drift slightly on OmnibusF4SD with MPU6000

wuhao19920619, 2018-02-24:
After I set the moron threshold to 200 it still occur sometimes that Gyro drift a lot after plug in with power. The indication in configurator just can't stop spinning and rolling. Which does not happened in 3.2.1.

Another one is Accelerometer drift slightly. It occurs every-time after I fly several seconds. The pitch angle drift +0.1 degree every 2 second and usually stop at about 4 degree. I guess it is not about the vibration because after flying for several seconds then I disarm and connect to configurator, but the drift still continue.

etracer65, 2018-02-24:
If anything setting the moron_threshold higher will make matters worse.  Leave it at defaults and be sure to let the quad be still during calibration to get the best result.  Also consider using `gyro_cal_on_first_arm = ON` to force a clean gyro calibration immediately before the first arming.

brucesdad13, 2018-02-24:
I am also experiencing tremendous acc drift on MPU-6500 and 9250 on more than 10 fcs with the more recent 3.2.x and 3.3.x. set imu_dcm_ki = 50 sometimes helps. I can lock the fcs into a vice on the bench and the virtual quad in the configurator is rolling on and on. Now and then if I replug the acc is rock solid but if I then move the fc a bit it starts drifting and never stops. Some drift in pitch some in roll... some in heading.. what type of data would be helpful?

wuhao19920619, 2018-02-24:
@etracer65 Thanks for your reply. I am just confused no matter set moron_threshold smaller or larger it will both make matters worse right?

wuhao19920619, 2018-02-24:
@brucesdad13 sorry to hear that. Just my opinion it sounds like hardware problem.

wuhao19920619, 2018-02-24:
@etracer65 I just test the betaflight 3.2.1. It can calibrate gyro pretty well during some movement. But with 3.3. After the calibration it will just move to the opposite direction of the movement during the calibration.

etracer65, 2018-02-24:
@wuhao19920619 Any movement during gyro calibration compromises the quality of the calibration and the gyro will be more subject to drift.  The `moron_threshold` sets how much movement is allowed during calibration for it to succeed.  There is inherently some noise in the gyro reading so some range must be okay.  Increasing `moron_threshold` so you can get a calibration while moving the quad is not desirable.  Movement = less accurately calibrated gyro.  Just plug it in and set the quad down until the calibration completes.

wuhao19920619, 2018-02-24:
@etracer65 So do you mean that if I set moron_threshold to 0 then no movement is allowed during the calibration. Therefore the calibration will only done after I put it still? So with 0 I should get the prefect calibration right?

etracer65, 2018-02-24:
No.  If you set it to 0 then it will never calibrate because every gyro has some noise.  For MPU6000 just leave it set to the default `set moron_threshold = 48`.

wuhao19920619, 2018-02-24:
@etracer65 OK thank you! So should I close this issue? The accl drift issue is still there. It basically another issue.Should I close this and put another issue about accl dirft?

etracer65, 2018-02-24:
Yes, I would close this and open one specifically about the accelerometer drift if you can repeat and supply steps to replicate.

closed 2018-02-24 by wuhao19920619