betaflight/betaflight #5303: Betaflight 3.3-RC2 does not boot with old SDcard

wuhao19920619, 2018-02-24:
My target is OmnibusF4SD.

I used to use old SD cards (both 512MB and 2GB) and both of them work fine with betaflight 3.2.1. but with this version if I install the 512 MB SD card the beeper just keep ringing and the system does not boot, even can not be connected to configurator. And for 2GB SDcard it just stop before the 3 beeps which indicate the gyro calibration. And the same can not be connected to configurator. Both of them have been tried to formatted with FAT and FAT32 and none of them works.

MajkCro, 2018-02-25:
Can confirm exactly the same problem with 1Gb SD card. Using SC Card Formattter, and target is OmnibusF4SD.

DieHertz, 2018-03-01:
Looks like my case, an old 2GiB card also stopped working on F4 in 3.3. Works on F7 though. I will debug it tomorrow, no worries (yet) :-)

davidoff99, 2018-03-02:
I have the same issue with 1GB SD card on Matek f405 OSD in 3.3.
No problems in 3.2.5 version.

DieHertz, 2018-03-02:
Thanks, I'll look into it soon.

davidoff99, 2018-03-02:
Maybe this is causing the problem: #4659 because when I power up my FC the led indicator is stand still or sometimes doesn't light up and it should always blinking because my transmitter is turned off.

hundemann, 2018-03-15:
I can confirm for Matek F405 CTR (thanks for putting me on the right trail); 
BF 3.3.0: non-working with old 2 Gb card, working with new 32Gb SD card (or without card inserted)
BF 3.2.5: working with old 2Gb card

DieHertz, 2018-03-15:
No solution yet, problematic commit found but it's not the cause of this issue, probably need some additional code for older cards. Hope to work on it this weekend again.

wuhao19920619, 2018-03-19:
Sometimes the FC can boot normally ,but only about 1/10 chance. If the FC boot then accessed with configurator, the blackbox just say there is a fatal error with SD card.

Tbeal79, 2018-03-20:
Yup jb f4  has the same, but I'm not getting blackbox at all in my cli dump, and I get the fatal error every time I plug mine in.
If you need to see the dump just click the link that wuhao posted.
I talked to jb yesterday about it and he said it could me for my case a defective board, I showed him my dump and diff.

0crap, 2018-03-26:
Exact same problem with a MATEKF405 (MKF4) board.
Sometimes the board boots and sometimes it does not boot. (Solid blue and red LED's when stuck.)
When it boots I can go into configurator, the Blackbox tab always shows Onboard SD card "Fatal error Reboot to retry".
Using a micro SD card 2GB in size.
Tested with 3.3.0 stable amd latest Jenkins build 715.

It's working fine with 3.2.5 stable.
Reverting to 3.2.5 stable for now.

wuhao19920619, 2018-03-29:
My FC is the pink EXUAV F4 PRO V1 and I tried EXUAVF4PRO V3.3. The problem still occurs.

eried, 2018-07-03:
This is still an issue :/

Tbeal79, 2018-07-03:
Mine is still the same way as yours. I think mine also is the 3.2.5 that the sd card works on. I think someone told me that you have to use a newer sd card now for all of them, even tho I'm using the same one for my bff3 board with 3.3.3 on it.Dont know why this is. Hope it dont change when the 3.4 comes out

0crap, 2018-07-04:
@Tbeal79 This issue still exist, Yes you can use a work around by using a "new" SD card.
I sure hope they will fix this at the final release. A lot of unused "old" SD cards can't be used now for blackbox logging.

eried, 2018-07-09:
@0crap but what "new" means? Bigger than some size? For the blackbox it would be nice to use a microsd that I do not care to lose, like 512mb ones :/

0crap, 2018-07-09:
@eried To me it seems "old" is the SD cards you find in the bottom of a drawer. :-)
So "new" is like a recent SD card you got from the shop.
All we know for sure is that BF 3.2.5 is the last know good with "old" SD cards.

Drak0ri, 2018-07-28:
Same here ... Two old cards one Fat16 2GB and one Fat32 32GB Both say error.... Both stop Betaflight from connecting when plugged into quad 90% of the time .... have seen them working 1 or 2 times only.

DronematesFPV, 2018-08-19:
running 2 quads 2 types of SD 8gb and 32gb only 6months old . 2 diff FC , 2 diff versions 3.4 and 3.5 both have fatal error on betaflight blackbox. So its Betaflight.

wildzcardz, 2018-09-12:
I was having the same issue.  What I had to do to get it working is I am on Win 10 and it does not let you do fat32.  Downloaded a program called Rufus Portable and made my sd card fat32 and now have it working for blackbox.  Also running the new HW stack with omnibusf4sd firmware.

0crap, 2018-09-14:
I've a 2GB sd card and Win 10 has no problem in formatting it using FAT32.
So I did that and gave it a try.
Still the same problem, FC freezes on boot.
(Sometimes the FC does boot and blackbox can log correctly.)

wildzcardz, 2018-09-15:
0crap, try a class 10 and bigger card like a 16gb.  I have a older 4gb that would not work.

naimo, 2018-09-15:
Same issue here with betaflight 3.5.2 OMNIBUSF4SD on a omnibus F4 pro V2 flight controller.
Tried both a 2GB (SD-formatter FAT16, or FAT32) and a 8GB class 10 (SD-formatter FAT32), and the SD card is not recognized (Fatal error Reboot to retry)

DronematesFPV, 2018-09-15:
I have a omnibus F4 pro V2 flight controller. And a CLs racing V1.4 FC both display fatal error. I went and bought a new sankdisk ultra 16gb . This is the one joshua bardwell uses and recommends. both display fatal error . Reflashing does nothin, both have Blackbox active SDCARD in CLI. I conclude betaflight has a floor and is not configured for SDCARD blackbox I believe.

DronematesFPV, 2018-09-15:
> Sure, that will work.
> But that is not the point, this bug is about old SDcards keeping the FC
> from booting properly.

Well its also happening with New cards too.:)

0crap, 2018-09-17:
@DronematesFPV Wow, that is really bad.

As far as I know there are only bug reports with old SD cards.
Sure hope it gets some attention from the devs.
Nasty bug it is.

iespoto, 2019-02-08:
Same issue here with
**Matek F405-CTR**
At first i got Fatal Error, later i changed baud rate for GPS and it didn't wanted to boot, so i thought that GPS is messing, disconnected it, still no luck - with boot went into DFU, reflashed, still couldn't connect. Removed SDcard - and everything worked, tried three SD cards that i had, and previously worked on Omnibus F4 ProCorner - no luck.
BF 3.5.5

iespoto, 2019-05-03:
Any resolution on this one?

nmare99, 2021-04-15:
Same issue here with
Matek F405-CTR BF 4.2.0 and Matek F405-CTR BF 4.2.3. FC boots, but "SD card is not recognized (Fatal error Reboot to retry)" in configurator. After switching to CLI and back to configurator, the SD card is detected! After power off and plug again - "SD card is not recognized (Fatal error Reboot to retry)".
PS. SD card 4Gb SDHC.

haslinghuis, 2022-05-08:
Closing as inactive

closed 2022-05-08 by haslinghuis