betaflight/betaflight #5304: Betaflight 3.3-RC2 Accelerometer drift slightly on OmnibusF4SD with MPU6000

wuhao19920619, 2018-02-24:
It occurs every-time after I fly several seconds. The pitch angle drift +0.1 degree every 2 second and usually stop at about 4 degree. I guess it is not about the vibration because after flying for several seconds then I disarm and connect to configurator, but the drift still continue.

etracer65, 2018-02-24:
Probably related.  https://github.com/betaflight/betaflight/issues/4573 - long read.

wuhao19920619, 2018-03-04:
@etracer65 in my case if I do the bench test everything looks okay, but after flying then do the bench test again without restart will see a significant drift.  In sensor tab it look pretty stable for accl data, and gyro have some inherent vibration, about +-1. I guess maybe somehow FC think accl is done then it just bypass the accl and let gyro to simulate accl. Is there anyway I can set this threshold? Or can this be adjusted like it will keep assessing the accl, once the error lower then threshold then the accl will come back to work.

etracer65, 2018-03-04:
If you're seeing +-1 on the gyro traces in the sensors tab when sitting completely still on the bench, then that's a lot of noise.  The gyro data is used as part of the attitude estimation so lots of noise will cause drift.

When connected only via USB (no battery) do you see the same level of noise in the sensors tab?  I suspect not since you indicated that the drift was not a problem when bench testing.

If the noise increases when you connect a battery that indicates that you need more power supply filtering and adding a capacitor may help.  I could also just indicate that the on-board power supply is just not very good - particularly the 3v3 rails.

If the noise is the same between USB vs. battery power, then possibly you just need more gyro filtering configured to allow less noise to get into the PID controller and IMU calculations.