betaflight/betaflight #5305: Investigate / fix partial config reset through parameter groups.

mikeller, 2018-02-24:
As identified in #5289, the current way that parameter groups are implemented is that if a parameter group version changes during a firmware upgrade, the values for this parameter group will be reset to defaults, but the rest of the parameter groups will keep their values.

This is well-meaning, but I think it introduces a problem in terms of user experience: If the parameter group in question had only 1 or 2 non-default values, it will be very hard for the user to look at a config `diff` or `dump` and spot that something that shouldn't change has changed, potentially causing a not correctly configured craft to be flown.

As suggestions for how to improve this:
- make resets of one or more parameter groups force a reset of the entire configuration, causing it to fail obvious;
- add a `config_needs_review` flag that will indicate that the config has been changed, and need to be reset before the craft can be armed.

etracer65, 2018-02-24:
I'd vote for the first option.  It would make it obvious that they need to examine and reset the configuration.  Consistent with the the EEPROM version changing.  People generally expect their config to get reset when upgrading anymore.

With the second option the problem of finding the reset values still remains.  Plus it would lead to more "why can't I arm" questions.

etracer65, 2018-02-24:
As a general improvement, maybe display a popup upon the first connection with the configurator whenever the settings have been reset.

AndersHoglund, 2018-02-24:
Simple solution, always reset settings during an update. 
As we can not guarantee settings will be preserved in all cases, lets turn it around and not even try, and state that clearly: "No settings will be preserved. Restore settings from a saved cli diff". In most cases the user has to do this anyhow. Make it valid for all cases. Preserved settings is a special case, and I hate special cases. Love normal cases.

mikeller, 2018-02-25:
@etracer65: I think you are right, resetting the entire config every time a parameter group changes is probably the right thing to do - 'doing it right' requires the user to have a backup anyway, and starting from a clean slate is what the users are already used to.

The drawback of this is that this will wipe a user's config even if the parameter group that got updated was all set to defaults, and did not have any user specific changes.

The second option will not help with this either - having to examine a diff line by line to find out if anything was lost isn't a good strategy, and restoring from backup will almost always be preferrable, even if nothing has changed.

@AndersHoglund: One problem with this is how to define an 'update' at the firmware level - is it every change of the commit hash that is baked in during a build? This would become very annoying for developers.
Also, there are lots of legitimate updates (like almost all patch releases) that should not require any config reset since changes requiring one should not go into patch releases (with the exception being safety updates). If we can leverage the parameter group versioning for this then we get this in an automated fashion. In general, I think we should try to make our software as user friendly as feasible.

mikeller, 2018-02-25:
@etracer65: Re popup, good idea. Could be as simple as a `configuration_reset` boolean parameter that defaults to `true` and is set to `false` as part of saving the configuration.

mikeller, 2018-06-02:
Implemented in #5991.

closed 2018-06-02 by mikeller