betaflight/betaflight #5306: Strangeness in MPU9250 gyrl DLPF configuration

etracer65, 2018-02-24:
I think this needs to be fixed, but I wanted to post the findings here to see if anybody knows why this was done before I prepare a PR.

**Background**:

For gyros **not** running in 32K mode, the `gyro_lpf` setting is used to configure the DLPF in the gyro.  The default setting is `gyro_lpf = OFF` (0).  The selected DLPF cutoff is set by applying the `gyro_lpf` value (0-7) to the `DLPF_CFG` portion of the register as shown in this table.

[image]

The complete enumeration for `gyro_lpf` is:

```
static const char * const lookupTableGyroLpf[] = {
    "OFF",
    "188HZ",
    "98HZ",
    "42HZ",
    "20HZ",
    "10HZ",
    "5HZ",
    "EXPERIMENTAL"
};
```

So the default of `gyro_lpf = OFF` corresponds to 0 which actually selects the 250Hz cutoff (the concept of "OFF" is not really true).  Note that the listed frequencies in the enumeration are not really correct as they're based on the MPU6050 gyro (every one has slightly different values).  Also note that "EXPERIMENTAL" (7) selects the high frequency cutoff of 3600Hz which will be considerably more noisy.  Lastly, any value for DLPF_CFG except for 0 and 7 will put the gyro in 1KHz sampling.  The `gyro_lpf` value is used directly as the DLPF_CFG value in the register.

So this is how it's supposed to work and the code follows these conventions when initializing the gyro in every case - **EXCEPT FOR THE MPU9250**

Somebody intentionally did something very strange in the code which doesn't make any sense to me.  `/drivers/accgyro/accgyro_spi_mpu9250.c`

```
    if (gyro->lpf == 4) {
        mpu9250SpiWriteRegisterVerify(&gyro->bus, MPU_RA_CONFIG, 1); //1KHz, 184DLPF
    } else if (gyro->lpf < 4) {
        mpu9250SpiWriteRegisterVerify(&gyro->bus, MPU_RA_CONFIG, 7); //8KHz, 3600DLPF
    } else if (gyro->lpf > 4) {
        mpu9250SpiWriteRegisterVerify(&gyro->bus, MPU_RA_CONFIG, 0); //8KHz, 250DLPF
    }
```

Just let that soak in for a moment...  It makes no sense based on the expected valued for DLPF_CFG.  Also the default `gyro_lpf = OFF` will cause this code to select the "EXPERIMENTAL" setting (7) and put the gyro in the less filtering/high-noise configuration.  And what's so special about the value 4 in this context?

I have no idea why this was done.  I looked back through the github history as far back as I could back to April 2017 and this code was there.  So as far as I know this goes back a long way.

So unless I'm missing something obvious I don't know why this was done and it seems like it really needs to be fixed.

etracer65, 2018-02-25:
@hydra I think the SPRACINGF3EVO was one of the first to use the MPU9250 (maybe mistaken on this).  Any thoughts on this issue and why the DLPF configuration is so unusual?

hydra, 2018-02-25:
no idea, ive nit seen that code before.

also i was against the gyro_lpf being using as the written to register. previously there was a switch statement which disconnected the user setting from the device driver.  iirc it was code from @borisbstyle

I'm in favour of any fixes to this area of code.

PR #5483 (2018-03-22): Rework gyro sample rate and DLPF configuration and expose additional filter cutoffs
Fixes #5353 
Fixes #5306 

The old `gyro_lpf` setting was based on the DLPF_CFG values for the MPU6050 gyro and the enumeration was inaccurate and misleading.  For example, the default "OFF" setting did not disable the DLPF, but actually set it to around 250hz.  The actual cutoff frequency for each setting varies by gyro hardware so the literal frequencies in the enumeration were also incorrect.

Removed `gyro_lpf` and replaced it with `gyro_hardware_lpf` (8KHz) and `gyro_32khz_hardware_lpf` (32KHz).  The param [... cut]
closed 2018-03-22 by mikeller