betaflight/betaflight #5310: Betaflight 3.3 - RC2 - RSS

stu22uk, 2018-02-25:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide information about your flightcontroller and other components including how they are connected/wired;
4. Add the used configuration and firmware version;
5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6. Remove this Text :).

stu22uk, 2018-02-25:
Updated both my CL Racing F4 FC's (identical setups) from 3.3.2 to 3.3.0 RC2 and pasted original config via CLI. Both had same issue with RSSI feedback as follows - 

- Only showing 12% in 'setup' tab despite reporting correctly to Taranis.
- Value did change with range during flight (displayed in OSD) but never went above 12%
- Reverted back to 3.2.5 and all was well again.
- Additional issue with OSD -  'mah' value normally in bottom right disappeared despite being selected as 'on' and showing in correct screen position in the 'OSD' tab. Again, reverting back to 3.2 fixed the issue.

Hope the above helps.

etracer65, 2018-02-25:
Next time please read the instructions for creating new issues.

Try using "full chip erase" when upgrading to 3.3RC2 and setting up the config manually - do not paste in a dump from 3.2 as a number of parameters have changed.

stu22uk, 2018-02-25:
First time - did my best to follow - 

_"If you believe there is an issue with the firmware itself please follow these steps:
Describe your problem;
Include ways to reproduce the problem;
Provide information about your flightcontroller and other components including how they are connected/wired;
Add the used configuration and firmware version;
Create a diff and post it here in a code block. Put ``` (three backticks) at the start and end of the diff block (instructions on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
Remove this Text :)"_

Disregard as you please, just trying to help.

etracer65, 2018-02-25:
@stu22uk I was just trying to help you with future issues, not disregard your question - plus I said "please" :)

Did you try my suggestion?

> Try using "full chip erase" when upgrading to 3.3RC2 and setting up the config manually - do not paste in a dump from 3.2 as a number of parameters have changed.

stu22uk, 2018-02-26:
Guys, 

Please consider that I didn't follow specific protocol not because I'm lazy but rather at the limit of my understanding. As a laymen I tried to feedback with what I could to assist the community. 
If rigourous procedure is preferred over a best attempt at community feedback then perhaps I'm not ready for this site yet.

@etracer, I had done full chip erase when updating. I noticed after pasting the dump from 3.2 that some things were not configured correctly so I rectified them - 'rates' for example. 
The only two things that I couldn't solve were as above. 
Now I understand a bit better about the diff command I will reflash 3.3 and try again and drop the values here. The weather was too good yesterday not to go flying - after all, that's why we're all here....

wind0r, 2018-02-26:
@stu22uk  All thoese message werent about a "rigourous procedure" or "protocol". It is just about that with the given data there simply isnt a way to reproduce your issues or get other informations out of it.

We are just asked for the diff/dump so people can investiage :)

Since as i said above:
There are 3 (and more) different methos to get rssi to the firmware without knowing which one you used its plenty of time other people need to spend to reproduce it or may not even possible to reproduce it.
Same goes with the OSD Screen, without the Layout its hard to reproduces.

And the other "prodecure" just was about removing the starting text and type the issue there. I think/hope that you agree that the your first comment to this issue should be the starting post and not the current procedure text.

Help and informations is allways wanted :) but as you can see with all the message here we are spending plenty of time befor even can start to investigate, because of that those time wastes those realy easy steps got created. As you said we are all here because we want to fly, so if one person can spend 30 secs more and saves 20 others multply of hours of work i think the way to go is quite easy. especially when we know that noone is paid for this and only spending there freetime here

If you can explaine where those steps are rigourous i would be quite happy so we can improve it. We wanted those steps to be as easy as possible so everyone in the community can take part, but contributors still have some data/information so they can start helping and not wasting time trying to figure out what to do. Providing a dump/diff is so easy that everyone should be able to do so.

SkylionFPV, 2018-02-28:
I'm having the same problem, the rssi seems to be reading inverted, like instead of counting down from 99% it's counting up from 12%, worked fine in 3.2.5, I have crossfire, I set it up the way the way I saw instructed in youtube, using LQ as rssi, adc_rssi active in config tab and set to aux 8 in the receiver tab, only change I've been able to make was inverting the rssi in the cli changed it to 62% rather than 12, I guess my question is in my "get rssi" search there was an option names rssi scaling that was set to 41, would changing that up or down get my rssi back to reading normally or is there something else going on here?

mikeller, 2018-03-01:
@SkylionFPV: Enabling `adc_rssi` and selecting an RC channel for RSSI at the same time does not make sense - you can only have one source for RSSI active. Which one do you want to use?
Also, adjusting RSSI scaling will allow you to calibrate RSSI to the range you want.

markml3, 2018-03-25:
@stu22uk I had the same issue, where after the update to 3.3 my RSSI seemed to be in reverse on my OSD (Bardwell CLRacingF4), but my Taranis Qx7 reported the correct value.  In the CLI, I entered **set rssi_invert = on** and it solved my problem.  Now the OSD and the Taranis both report the correct value.

jumatic, 2018-03-25:
that doesnt help, anyway, setting up min/max like minimosd is the way. 
turning on/off TX within 15 seconds, done.

------------------------------------------------------------------------
Gesendet mit der Telekom Mail App
<http://www.t-online.de/service/redir/email_app_android_sendmail_footer.htm>

--- Original-Nachricht ---
Von: markml3
Betreff: Re: [betaflight/betaflight] Betaflight 3.3 - RC2 - RSSI 
Value/Scaling (#5310)
Datum: 25.03.2018, 7:57 Uhr
An: betaflight/betaflight
Cc: jumatic, Comment

@stu22uk <https://github.com/stu22uk> I had the same issue, where after the 
update to 3.3 my RSSI seemed to be in reverse on my OSD (Bardwell 
CLRacingF4), but my Taranis Qx7 reported the correct value. In the CLI, I 
entered set rssi_invert = on and it solved my problem. Now the OSD and the 
Taranis both report the correct value.

MathijsG, 2018-04-08:
I have the problem that it's inverted. When I have 97% signal, I see a 3% RSSI instead. When I use the invert parameter, it's still showing 3%. What can be wrong? Using an SP Racing F3 and a FlySky I6.

markml3, 2018-04-08:
Did you enter "save" in the CLI after you entered "set rssi_invert = on"? Did you remove all power from the quad and then power it back up? 

I had the same issue with both my quad and my micro quad after upgrading to BetaFlight 3.3 and this setting solved the issue for both, but I had to save and power cycle before I saw the change

MathijsG, 2018-04-08:
Sorry I forgot to save. Excuse my noobism :-P

closed 2018-05-15 by stale[bot]

TMac54, 2018-05-19:
@markml3 :  So, you're saying that in BetaFlight 3.3 if you used to have set rssi_invert  = OFF, NOW it needs to be set rssi_invert = on ?  I just installed an XM+ receiver (which used to require rssi_invert = off in Betaflight).  I am running BetaFlight 3.3.1.  XM+ receiver does not give telemetry only RSSI on channel 16 so I don't have RSSI values in my Taranis QX7 to check against.  The only thing I have is BetaFlight RSSI value and OSD RSSI value.  Both of these values match each other but both are only showing a value of 10.  The RSSI signal appears to be working fine on channel 16 (Aux 12) in Betaflight.  Quad seems to be responding to all TX commands (on the bench--not flying yet as I ahven't put it back together once I saw RSSI value of 10). Of course, when I set rssi_invert = on  and save in CLI it DOES now show an RSSI value of 90.  I am confused however as to whether or not I am seeing the correct RSSI value ofr just "masking" a problem.  Did Betaflight dev's put out a version 3.3 (and above) update on thsi supposed new "twist" to the set rssi_invert = ON/OFF command for XM and XM+ receivers?  Did I do the right thing and can I trust this new "90" value? Thankyou for your time.

markml3, 2018-05-19:
I haven't seen anything from the devs on this, but I haven't looked that hard. Since it solved my problem I just went with it. I have 2 quads that need it inverted and one that doesn't. No idea why. 

If you're not sure if it's working or not, while everything is on, but the quad isn't armed, take the transmitter and walk away from it. See if the RSSI behaves as you would expect. If it does, you're good. If not, you've got other issues.

TMac54, 2018-05-19:
Thank you for your response.  I’ve looked and have not found anything.  I have asked Joshua Bardwell and he said he didn’t know off the top of his head but to “experiment.”
Guess I am in the same boat as you.  The “No idea why” thing really bothers me (I’m the type  that really wants to know why).  Guess I’ll just need to get over it!
Thanks again for your time and posts!

mikeller, 2018-05-19:
As to why: There's a bunch of technologies that are used for the analogue RSSI signal. Most RX don't actually output an analogue signal, but PWM corresponding to an analogue signal. Some flight controllers have an RC element on the RSSI in to convert the PWM into analogue before feeding it into the ADC, others use an opamp for this, and some have the ADC directly connected to the RSSI pin. In the 'opamp' variety, there are some that are clipping if the voltage level is too high. This means that every RX / board combination will need to have the right settings for analogue RSSI to work.
As a workaround, I'd ditch analogue RSSI and go for a protocol that has RSSI baked in, like FrSky FPort, Crossfire, or the newer Spektrum variants.

markml3, 2018-05-19:
Good info. Thank you. My only question is why did the behavior change when I updated to BF 3.3.x? Not a big deal. As long as I can get it to work I'm good. I just found it interesting.

TMac54, 2018-05-19:
Thank you for the info

mikeller, 2018-05-20:
@markml3: This wasn't a planned change - it is unintended, and most likely caused by some changes in how ADC or RSSI are handled.

will160, 2018-11-01:
I am having same issue. Rssi starts at 12%. Seems to have good signal just reads wrong. It is on all my quads which makes me think it is something i did wrong in my tarranis or maybe with the crossfire. I did have some issues and had to reset everything but they were unrelated and i think was already doing this. It even does it with a new flight controler i did not update.

jumatic, 2018-11-01:
Please use newest firmware. The rssi value is adjustable also reverseable. 
Sounds like in your case it is reversed. set rssi_invert = on
If raus doesnt work something else could be the prob. Maybe try more 
distance between rx tx

------------------------------------------------------------------------
Gesendet mit der Telekom Mail App
<https://kommunikationsdienste.t-online.de/redirects/email_app_android_sendmail_footer>

--- Original-Nachricht ---
Von: will160
Betreff: Re: [betaflight/betaflight] Betaflight 3.3 - RC2 - RSSI 
Value/Scaling (#5310)
Datum: 01.11.2018, 3:37 Uhr
An: betaflight/betaflight
Cc: jumatic, Comment

I am having same issue. Rssi starts at 12%. Seems to have good signal just 
reads wrong. It is on all my quads which makes me think it is something i 
did wrong in my tarranis or maybe with the crossfire. I did have some 
issues and had to reset everything but they were unrelated and i think was 
already doing this. It even does it with a new flight controler i did not 
update.