betaflight/betaflight #5311: Betaflight 3.3.0 RC2 - Barometer sensor gives sporadic readings

andyprice44, 2018-02-25:
I just wanted to make aware that flight controllers with a barometer may have sporadic readings in 3.3.0 RC2. I am currently using a KakuteF4V2 and the altitude reading using 3.2 was correct. While using 3.3, as soon as you plug the battery in the reading starts ascending rapidly. Once you take off, the reading becomes sporadic. This happens on every flight.

ttjs82, 2018-02-25:
I noticed the same issue with the KakuteF4V2 and Betaflight 3.3.0 RC3 full chip erased. Just to make sure I tried flashing the Betaflight again, but the issue persists. However, instead of ascending in my case the barometer altitude reading seems to start decreasing after plugging in the battery. This can be seen both on the Betaflight sensors tab and through the altitude telemetry reading on QX7S transmitter. I haven't tried to fly this quad yet.

etracer65, 2018-02-25:
If it behaves properly while plugged into USB but only acts up when running off battery, then it's likely the board's power supply is noisy or not very stable and this is affecting the sensor.  Another possibility is that the on-board regulators are close enough to the baro sensor and are generating heat which would also affect it.  These problems wouldn't manifest when connected via USB since the power comes from the USB port.

ttjs82, 2018-02-25:
Sorry, to be accurate I'm seeing the issue also while connecting just with the USB and watching data on sensors output tab for the barometer. The number for the barometer keeps decreasing continuosly. Is there information of the location of the barometer sensor on this board @etracer65 ?

ttjs82, 2018-02-25:
@etracer65 seems to be correct that the heat could be the culprit for the issue as although the barometer reading seems to continue decreasing also with just the USB connection it's much slower than with also battery connected at the same time. I went back to Betaflight 3.2.5 to check if the same happens with that version, and it does happen.

Perhaps there could be some kind of correction based on temperature reading (if available through the processor or some other sensor) to make the barometer more immune to the heat messing up the reading but this isn't that big of an issue on an fpv quad. For some autonomous flight quad it would be more critical to have more accurate altitude reading but then again external sensors instead of integrated ones would also make better working build in that situation.

So for me this issue seems to be something we just have to live with.

etracer65, 2018-02-25:
One thing to remember is that the baro sensor is not really super accurate.  The BMP280 which is on the Kakute is only rated for +- 1M of accuracy.  Normally this is fine for flying but look at it on the ground when not moving can appear to be a lot of drift when it won't really matter in flight.

There actually is a temperature sensor in the chip to help adjust because the sensor is so sensitive to heat but it only does so much.

So the reality is that a slow drift from the baro output is simply unavoidable with the low resolution device used on most flight controllers.  The BMP280 wasn't made for drones, but originally targeted at consumer devices like phones.

andyprice44, 2018-02-26:
While it is true that dirty power and/or heat may play a factor in sporadic readings, I will say that the baro gave pretty close to accurate readings while using betaflight 3.2. It would normally read 5-7 feet when sitting on the ground but, it was stable. When it flight it would read close to the actual altitude of the craft. When I flashed 3.3.0RC2, this is when I began having the problems. Now the reading will rise 50-100 feet per second upon plugging in the battery and the craft is on the ground. When in flight the readings are very sporadic and can't be used.

ttjs82, 2018-02-26:
I've have gone straight to the Betaflight 3.3.0 RC3 as this is a fresh build and don't have extensive comparison of 3.2.5 with this FC. I might have to do some kind of comparison and logging to see the real difference.

[image]

I did locate the barometer sensor on the board. I've the board mounted as low as possible and clearance between the pdb and frame is 4 millimeters and some of the components extending from the board are even closer. The baro sensor is facing the frame and even the frame does heat up quite a lot whether powered through USB or LiPo, maybe I could get a infrared thermometer to get actual measurements or try logging the temperatures with blackbox between the Betaflight versions. This heat buildup was actually one of my concerns from the get go as there are at least the main processor and OSD chip radiating heat on the bottom of the board and perhaps suffering from the same heat. I might just raise the stack a little bit to give it more breathing room. The power components on this setup should be quite interference clean as I use Tekko32 ESCs and also an additional capacitor on the power lead.

Did you @andyprice44  make any component or setting changes when going from 3.2 to the 3.3.0? Maybe some changes in the settings or the Betaflight 3.3.0 itself just make the FC heat up more compared to the 3.2 thus making the sensor readings more erroneous. Did you try the release candidate 3 yet?

etracer65, 2018-02-26:
I only have one board with a barometer, but it happens to have a BMP280.  Though not a Kakute.

Testing with BF3.3 shows nothing unusual.  Actually I was surprised how stable the baro output was after over 10 minutes.  No persistent drift, just the expected noise variations around zero.  Here's the sensors tab showing 5 minutes of readings (1 second update).

[image]

After about 15 minutes it seemed to stabilize at around -1 meter but that's well within the tolerances expected and probably due to heating drift.  Note that this board was bare on the table and not mounted in a frame   The board was powered from USB so it minimized any effects of heating from on-board regulators (which would likely be bad on this board as the main inductor is mounted on the opposite side of the board directly below the baro - AirbotF4 Dual-gyro).

Some other things to note regarding baro sensors:

- They are very sensitive to wind.  The normal practice is to cover the sensor with a block of open-cell foam to minimize the effect.  Air flow from the props into the frame can really affect the readings in flight.

- Believe it or not, the baro sensors are also light-sensitive (try shining a flashlight on it and watch the result).  You must ensure that the sensor is not affected by sunlight.  Again the open-cell foam can help but it needs to be opaque.

ttjs82, 2018-02-26:
KakuteF4V2 AIO seems to take a lot more time to level out even with just USB.

[image]
this is from 0 to 5 min period after connecting just the USB no lipo when the board was cold (room temperature). After 10 minutes the reading is -12 meters where it seems to level out.

After the 10 minutes period I quickly disconnected the board and reconnected. Now because the board was already warm from the previous 10 minute measurement the readings from the baro were a lot more stable. After 5 minutes the reading was this time -1.5 meters with and quite stable.

[image]

These findings agree with your theory @etracer65 of the heat effect. I think I'll just set up a logical switch to reset telemetry or at least the altitude reading while arming the quad to get more accurate starting altitude and even without the -10 meters error is something I can live with. Maybe I'll even set an default offset of 10 meters in the Taranis if the error seems to repeatable. But I guess there isn't an offset setting in betaflight?

mikeller, 2018-02-27:
@ttjs82:

> I guess there isn't a way to reset baro reading while arming in betaflight?

Not yet. But it can be added - probably best as an option (potentially 'off', 'on', 'first time only').

ttjs82, 2018-02-27:
@mikeller Those options would be really helpful.

I found out that there was this CLI variable baro_hardware = AUTO by default. I changed it to the BMP280 based on the chip on the Kakute just to see if the issue was perhaps because of an error in recognizing the chip automatically. However, this didn't change help as I still got consistently the ~10 meters drop in the readings when plugging in the board to the USB the first time. With Lipo the drop happens even faster so Mike's suggestion of the Betaflight options for enabling the barometer reset while arming ('off', 'on', 'first time only') would allow for more meaningful baro readings as this way it would have some time to stabilize between plugging in the power and takeoff.

jflyper, 2018-02-27:
Wait. 10m drop at power up is unusual.

ttjs82, 2018-02-27:
yes @jflyper check the images https://github.com/betaflight/betaflight/issues/5311#issuecomment-368524013

jflyper, 2018-02-27:
Hmm... I was expecting much more rapid drop.

Is this repeatable as many times as you want?

10m worth of change (roughly equal to 1 [hPa]) over 10min period ... can natural transition from low to high pressure zone this fast?

To evaluate this kind of stability, we need an air-tight bottle or something, don't we?

ttjs82, 2018-02-27:
Yes, seems to be repeatable @jflyper. I tried to repeat this today and the results seem to be the same. This 10 meter drop seems to happen consistently. 5 minutes to -10 meters and then at 10 minutes about -12.5 meters. After 15 minutes it still at -12.5 meters. The cpu_overclock setting is off.

This from today 0-5minutes:
[image]

Seems like the heat generated through the usb connection and the heat dissipated to the air even out at around 10 minute mark with the reading -12.5 to -13 meters.

10-15 minutes:
[image]

And then disconnecting the usb and plugging in back as quickly as possible (so the board has now been warmed by the previous 15 minutes connection):
[image]
This time the reading stabilizes between -1.5 and -2 meters if I wait for 10-15 minutes from plugging in the USB. The ~0.5 meter max spread of the readings is in my opinion pretty good once the readings stabilize.

With Lipo connected the -10 meters is in about 75 seconds and the readings continue to drop to about -24.5 meters before stabilizing again 10-15 minutes from the start:

[image]

This is definitely interesting and naturally it would be nice to be able to get somewhat accurate measurements from the baro. However, for traditional fpv freestyle or racing the altitude readings doesn't matter that much. Still it's no excuse for not trying to improve this thing. And I guess there could be situations where you would have to prove authorities that you haven't broken the maximum allowed altitude ceiling and in that case having more accurate measurements would prove helpful.

ResinSmoker, 2018-02-27:
The barometer is just a cavity with a capacitor and an open hole on one end or the other. The difference in pressure is felt by the cap changing it's value. Unfortunately, exposing the Baro to the open air causes it to experience turbulence which can result in the erratic readings. (Can cover up with a small bit foam plastic to act as dampener)

Furthermore, the Barometer casing appears to be glued on as I've had one pop off after heat shrinking wiring nearby. It's possible that exposure to heat can cause one of the casing ends to loosen, resulting in air entering the cavity. Thus preventing it from reading correctly.

pitts-mo, 2018-04-06:
@mikeller should separate issue be created to add feature/setting mentioned by you and @ttjs82 ?
add cli setting to allow reset of barometer on arming ('off' (default),'on','first time only')
thank you

more info:
My new Kakute F4 AIO V2 also drifts a good 10-13 meters over several minutes after cold start. I also found this interesting adafruit post where user sees BMP280 drift and compares it to LPS25H. 
https://forums.adafruit.com/viewtopic.php?f=19&t=122117

mikeller, 2018-04-06:
@pitts-mo: Good point, yes please.

pitts-mo, 2018-04-08:
Please provide feedback/direction.

Evaluating my Kakute altitude estimates yesterday lead me to believe creation of two new issues would be better:
**1. add support for manual initiated BaroCalibrationCycle()*? ( mode switch, OSD, Stick command )**
**2. add cli setting baro_cal_on_first_arm similar to gyro_cal_on_first_arm**

And possible change to aid understanding baro anomalies and integration issues as suggested in #5311:
**3. add output of raw barometer pressure and temperature data somewhere ( for debug and user integration optimization )**

background:
My Kakute altitude estimates yesterday were consistently drifting negative a few meters in the minutes after power up and then drifting positive the same amount plus a few more meters within the first minute of flight...  However once stabilized in flight I estimate the difference in altitude reported in betaflight OSD between value reported at ground level and whatever the current position was did appear to be reasonably accurate. Then on disarm the altitude estimate slowly drifted back toward the original negative value seen pre-arm.

test setup:
Betaflight 3.3.1 Kakute F4 AIO V2 
w/1x2x2cm piece of open cell foam sandwiched between BMP280 and QAV210 (similar results were seen in prior tests without foam)
starting altitude 317m
rather steady wind gusts up to 32kph (not enjoyable)
-6c to +3c ambient temp over several hours 
7 packs starting with an odd pack and then pairs over several hours ( quick power cycle will re-zero the pre drifted altitude value )

Speculation:
The predictable off,disarmed,flight deviations I saw yesterday appear to be more relative to temperature changes? As I figure altitude changes induced by a larger part pressure value change should have presented themselves quicker...  But wonder if this is an over simplified guess.

More Kakute tests:
I think I will try wrapping the entire stack with painters tape in an attempt to buffer temperature and pressure a bit more.

mikeller, 2018-04-08:
@pitts-mo: Yes, I think your suggestions make sense.

1. is  simple and straight forward, and can be done by adding `DEBUG_BARO` or similar to the list of debug modes, and then using `DEBUG_SET(DEBUG_BARO, ...)`  to log the raw values;
3. is also simple, and the template of `gyro_cal_on_first_arm` can be followed;
for 2., a stick combination would have to be defined, or a new switch assigned. Apart from that it's simple.

pitts-mo, 2018-04-14:
Today I flew 3.4.0 (2018-04-09) with performBaroCalibrationCycle(); tucked into disarm(void) and find the behavior helpful when I fly for a good 30 seconds and land before continuing... For my case I still think debug would be necessary to find root cause of altitude deviations.

closed 2018-05-21 by stale[bot]

telliott187, 2018-06-13:
are you seeing an i2c error?  I am noticing an i2c error.  I see the barometer uses i2c.  could be related.

pitts-mo, 2018-06-22:
@telliott187 Where are you able to see i2c error? What controller and build version?

I haven't had time to look into this myself but I believe there is still an issue in here somewhere. My very limited understanding of i2c data handling and the predictably negative drift a few meters prior to arm then positive many meters in first few seconds of flight was making me think thermal/pressure interactions caused by sensor location within my build (been wanting to add baro debug -distinct from altitude debug)

delagela, 2018-07-21:
I confirm this diag on two kakute aio V2.

On 3.4, long drift to about 10/15 meters on the bench

Revert to 3.2.4 no more drift on startup. ( I ve wait for the boards to cool down )

This seems to be related to thermal correction.

Im sure this started as soon as first 3.3 RC

Let me know if i could help on bisect.

Other prob, since 3.4 rc1, numerical Vario is Always null on Osd, but pretty sure that ils n'ont related

pitts-mo, 2018-07-21:
@delagela Thank you for clear presentation of this info.

**Is anyone else able to quickly verify these distinct baro behaviors were introduced between 3.3.0rc1 - 2.2.4 on kakute F4 v2 AIO?** 

undesired behaviors as I see them:
-a few meters of pre-arm negative drift that should be noticeable within 30ish seconds after power up
-then several/many meters of post-arm/in flight positive drift within 30ish seconds of flight

notes:
-The amount of drift does appear to be temperature sensitive.
-3.4.x attempts to re-zero altitude estimation on arm but you should still see the positive drift

more info:
 I was using fport (3.3.x) by the time I got first controller with baro. I will try to make time to test when this was introduced myself in the next few weeks if no new info surfaces.

etracer65, 2018-07-21:
I don't have a Kakute F4 v2 board but I just tested a Matek F722 which has the same BPM280 baro using BF3.4 and I don't see any problems.  Left it sitting on the bench for over 20 minutes and from power on it never varied more than 1m - withing the documented accuracy of the sensor.  The Matek F722 also runs quite warm and the baro is on the bottom so it was experiencing quite a large temperature shift.  During the test the baro went from 22C to 56C and remained perfectly stable.  So I'm not seeing any problems related to the code or driver for this particular baro sensor.

It's always possible that there are some design flaws with the Kakute boards that are maybe affecting the stability of the baro sensor, but I would have no way of telling that.

Substantial changes to in-flight readings are almost surely caused by moving air affecting the sensor - very difficult to prevent this completely.  But again, whatever possible board flaw (maybe on-board power stability) might be amplified by the noisy electrical environment while in flight.

ResinSmoker, 2018-07-21:
> Testing with anything that blows air across the sensor is not a good way to confirm anything. The moving air will have a bigger impact on the sensor than any induced temperature change.
>
> One other thing to note is that baro sensors are light sensitive and ambient light or even worse sunlight can significantly skew the readings. For example I'm able to induce a 1.5M negative offset simply by shining a flashlight on the sensor on my Matek F722.
>
> So the general rule with MEMS type baro sensors is to cover them with opaque open cell foam that will block wind and prevent light from reaching the sensor. Although the foam is unlikely to do much in the vicinity of 4 props moving lots of air quite close to the sensor.

What you need is a chamber to both isolate the FC and to allow for the pressure to be manipulated in a controlled way. This would allow for all the testing you could ever want so long as it was just a bench setup. Maybe a mason jar could work if you could figure out how to run the wires through the glass lid and yet maintain a seal. After that, add one vacuum pump and precision pressure gage and your in business. BTW I’ve worked with similar setups in the past that are used to test and check survival radios. Thus it’s possible that a commercial solution exists that could be acquired on the cheep.

NGFPV, 2020-03-09:
Has this issue ever been solved? I just built a Tricopter LR around a Kakute F4V2. My OSD reads altitude-=0 when I arm but it climbs more and more above true altitude as time passes. When I land, it is generally reads at least 160 feet as it touches ground at the original take-off site. I have tried both Betaflight and iNav/Triflight, and I have the same problem in both. Here is an example with iNav. My internet is impossibly slow, and I haven't succeeded at uploading a betaflight example. 
https://www.youtube.com/watch?v=00ktWhqdOe4&t=186s

etracer65, 2020-03-09:
@NGFPV You are commenting on an issue from over 1.5 years ago. There are no known problems.

If you have the same problem across multiple different firmwares then it would strongly suggest you have a hardware problem.

ResinSmoker, 2020-03-10:
One of the things to be aware of is that the Baro capacitive sensor is very touchy when air moves across it. Simply put, during flight the Baro is experiencing prop-wash, wind and turbulence over the frame. This airflow creates rapid changes in pressure that are mistaken for altitude changes. To combat this, add a small piece of foam over the Baro sensor to shield it. This will to some extent work but its better still to encase the entire FC in a shell or canopy if your wanting accurate data.

Beyond all this, if your Baro still misbehaves, its likely that the sensor is damaged somehow. Unfortunately this can happen if the FC was in a crash, got too hot during soldering or has FOD (Foreign Object Damage) in the sensor hole.

NGFPV, 2020-03-12:
I don't think the problem is prop wash. I have three Kakute F4V2 flight controllers - one in a quadcopter, one in the tricopter and one new in the box. The altitude reading drifts on all three just sitting on the kitchen table. I have an OmnibusNXT F7 that does the same thing. My Matek flight controllers read accurate altitude both indoors and in the air. I think for the Kakutes my solution will be to disable the barometer and depend on GPS for altitude.